OctoFarm/OctoFarm #288: [BUG] GridStack dashboard page tiles not flex-filling

davidzwa, 2021-02-14:
Developer report

GridStack looking weird in new webpack version (didnt check in og dev)

[image]

NotExpectedYet, 2021-02-14:
Will be same everywhere I think. Noticed it myself just never registered it so thanks.

davidzwa, 2021-02-14:
Just gotta make sure it's not my doing. Will come back with verification.

NotExpectedYet, 2021-02-14:
> Just gotta make sure it's not my doing. Will come back with verification.

Is it the new graphs not flex filling or them all? That printer progress shouldn't be allowed to drop to that size tbh.

davidzwa, 2021-02-17:
Not present (dragging problem was related to jQuery issues and webpack)

closed 2021-02-17 by davidzwa