OctoFarm/OctoFarm #298: [BUG]

Jayjayd123, 2021-02-15:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Describe the bug**
I cannot get to the printer setup page on octofarm

**To Reproduce**
Steps to reproduce the behavior:
1. Start octofarm
2. login
3. press setup printer
4. octofarm then takes me to the printer management screen and not to a place where I can setup a printer

**Expected behavior**
I expect to be taken to the page where I put my api key and my ip to add my octoprint instance running on a seperate raspberry pi on the network.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - NodeJS Version [2.0.7]
 - OctoFarm Version [1.1.11]
 - OctoPrint Version [1.5.3]
 - OctoPrint Plugins [None]

**Additional context**
Add any other context about the problem here.

NotExpectedYet, 2021-02-16:
Are you logged into the first user you created?

Jayjayd123, 2021-02-16:
> Are you logged into the first user you created?

Yes.

NotExpectedYet, 2021-02-16:
What does the printer management screen look like?

Jayjayd123, 2021-02-16:
> What does the printer management screen look like?

[image]

Jayjayd123, 2021-02-16:
it takes me to this page everytime I press "setup printers"

NotExpectedYet, 2021-02-16:
That certainly "looks" like your not login in with the admin user. That's what you'd see anyway. 

Could you pass me through the OctoFarm-API.log file please?

NotExpectedYet, 2021-02-16:
You should have another section to the left called Administration, you'd see add/edit/delete and others then.

NotExpectedYet, 2021-02-16:
Curious... Node version 2? Is that a typo? 

Also 1.1.11 isn't released yet so I gather you pulled from the development?

NotExpectedYet, 2021-02-16:
I've moved to the default branch back to master. There could be a bug on that development branch from recent work. I recommend switching back to master for now. 

`git checkout master` 
`git pull` 

Should sort you

Jayjayd123, 2021-02-16:
> I've moved to the default branch back to master. There could be a bug on that development branch from recent work. I recommend switching back to master for now.
> 
> `git checkout master`
> `git pull`
> 
> Should sort you
> 
> If it still persists let me know.

My node version was a typo lol! My Nodejs is actually running v12.20.2 and I did your suggestion to roll back to 1.1.10 and that did not solve the issue. I took a look at the OctoFarm-API.log file and it seems to be empty and I'm controlling through ssh but I can plug it in and grab the actual file if that helps.

NotExpectedYet, 2021-02-16:
Thanks for getting back to me. 

No errors if that's empty then.

Hmmm, I'll spin up a new install in the morning and see if I can reproduce. 

It seems like somehow your user didn't get the admin flag on creation. 

if you've got the pm2.error.log and pm2.log handy those might help but if I can reproduce tomorrow should be a simple fix.

Jayjayd123, 2021-02-16:
> Thanks for getting back to me.
> 
> No errors if that's empty then.
> 
> Hmmm, I'll spin up a new install in the morning and see if I can reproduce.
> 
> It seems like somehow your user didn't get the admin flag on creation.
> 
> if you've got the pm2.error.log and pm2.log handy those might help but if I can reproduce tomorrow should be a simple fix.
> 
> I think in my infinite wisdom I didn't do much loggin around user creation but I guess something got messed up in a recent release.

Awesome! I'll see if I can get those files to you tonight and thanks for the help. I'll be awaiting your reply.

NotExpectedYet, 2021-02-16:
No worries at all. All I can say is sorry 😅

davidzwa, 2021-02-21:
@Jayjayd123 we're waiting for your logs. Please be so kind to provide those in order to progress this issue.

Jayjayd123, 2021-02-24:
The pm2 error log is empty and I don't see a pm2 log in the logs folder.

davidzwa, 2021-02-24:
Can you open up developer console (F12 or click right and inspect in Chrome)
- Can you share any errors there?

Can you tell us about how you setup OctoFarm?
- Did you directly run it in NodeJS or did you use docker?
- How did you setup your MongoDB? Where is it running?
- How are you accessing OctoFarm (localhost/network/VPN/...)?
- Is your database from older versions of OctoFarm?

We'll use this to get started. You can also approach us/me on discord for faster help.

Jayjayd123, 2021-02-24:
> Can you open up developer console (F12 or click right and inspect in Chrome)
> 
> * Can you share any errors there after refreshing the page on Printer Management?
> 
> Can you tell us about how you setup OctoFarm?
> 
> * Did you directly run it in NodeJS or did you use docker?
> * How did you setup your MongoDB? Where is it running?
> * How are you accessing OctoFarm (localhost/network/VPN/...)?
> * Is your database from older versions of OctoFarm?
> 
> We'll use this to get started. You can also approach us/me on discord for faster help.

The first error is 

Uncaught TypeError: Cannot read property 'addEventListener' of null
    at Object.20../lib/functions/file.js (printerManagementRunner.min.js:9333)
    at o (printerManagementRunner.min.js:1)
    at e (printerManagementRunner.min.js:1)
    at printerManagementRunner.min.js:1

the second error is 

Unchecked runtime.lastError: The message port closed before a response was received.

I ran it using npm start in the octofarm folder via ssh over my network

My MongoDB start file path is /lib/systemd/system/mongod.service and I installed via the same method on the octofarm website matching the correct ubuntu version.

Octofarm version is 1.1.10

NotExpectedYet, 2021-03-05:
Can you update to the latest version please and start with a fresh database?

Jayjayd123, 2021-03-08:
> Can you update to the latest version please and start with a fresh database?

Eureka! I've reinstalled Ubuntu on my device and reinstalled OctoFarm with 1.1.12 and I've connected one of my printers so I think it's safe to say everything is working now. Thanks for all your help!

NotExpectedYet, 2021-03-08:
> > Can you update to the latest version please and start with a fresh database?
> 
> Eureka! I've reinstalled Ubuntu on my device and reinstalled OctoFarm with 1.1.12 and I've connected one of my printers so I think it's safe to say everything is working now. Thanks for all your help!

Epic to hear man! I gather it was the developer version we'd mistakenly made default at the time so apologies for that and happy printing! I will close now :)

closed 2021-03-08 by NotExpectedYet