OctoFarm/OctoFarm #299: [FEATURE IMPLMENTATION] Patreon: A configurable option to super charge the list view. Allow for thicker rows and bring in a camera stream and full action buttons.

NotExpectedYet, 2021-02-16:


brooksware2000, 2021-02-17:
[image]

Here is a mockup of what I thought could be done to consolidate camera, list, and panel views into a single screen.

There is a checkbox to enable or disable the camera view.  My thoughts are to have a placeholder panel in that location with a default cog graphic or something for instances where the camera is disabled or not installed.  Having a placeholder here could make it easier to align the controls without shifting them left if a camera is not available or right.

I added a new button on the screen where a user could load a print without opening the printer control interface.

I really don't see too much of a difference between the panel view and the camera view; hence, my recommendation is that all three screens be consolidated into a single view.

One final thought is to have an option in the system settings to set the camera view's default size with a max setting.

Thanks,

Curtis

NotExpectedYet, 2021-02-17:
Dude that's thorough! Thank you for that. The views are for different screen sizes/usages generally but yeah I do agree. 

I think I can manage that. 

I have two ideas for implementing. 
1. An option to enable a super charged list view, just replace the current. 
2. The option to disable the view menu for any of the links.  
So you could then just have this as your sole view and other users can also disable views they don't use. 

If that's acceptable I will take on the work.

brooksware2000, 2021-02-18:
> Dude that's thorough! Thank you for that. The views are for different screen sizes/usages generally but yeah I do agree.
>
> I think I can manage that.
>
> I have two ideas for implementing.
>
> 1. An option [...]

Yeah, I think your idea is spot on.  Looking forward to your updates as you have time to roll them out.

Curtis

NotExpectedYet, 2021-04-24:
No stale

davidzwa, 2021-07-31:
Eager to work on this power tool soon! The rework of OctoFarm took us to great ends

NotExpectedYet, 2021-11-06:
> 
> 
> [image]
> 
> Here is a mockup of what I thought could be done to consolidate camera, list, and panel views into a single screen.
> 
> There is a checkbox to enable or disable the camera view. My thoughts are to have a placeholder panel in that location with a default cog graphic or something for instances where the camera is disabled or not installed. Having a placeholder here could make it easier to align the controls without shifting them left if a camera is not available or right.
> 
> I added a new button on the screen where a user could load a print without opening the printer control interface.
> 
> I really don't see too much of a difference between the panel view and the camera view; hence, my recommendation is that all three screens be consolidated into a single view.
> 
> One final thought is to have an option in the system settings to set the camera view's default size with a max setting.
> 
> If you like this idea, I can work on some possible mockups for layout on smaller screens (thinking cell phones, etc).
> 
> Thanks,
> 
> Curtis

Done some work on this today, some screenies of what's coming below. Happy for input and tweaks! 

I'm working on a different set of Quick printer actions (they're in progress), but I have included your "Load" button as I really like the idea. How did you envision that working? I was thinking load could either pop up an upload box, or select previously existing files. I'm leaning more to a quick upload box than selecting due to the file manager having it's own button here now. 

[image]

[image]

Please let me know what you think! 

I left the camera enable/disable button selection off from the view because this is handled pretty automatically as long as OP is setup correctly. If camera url is available it will display, if no camera url or op has camera disabled, it will double check if thumbnail plugin data exists, if that doesn't exist then then the rows will fill the space. Thumbnail will fill the space if that exists. Maybe I'm missing the point of that button? 

Apart from that, I've currently re-used current styles and what not for information.

NotExpectedYet, 2021-11-06:
[image]

NotExpectedYet, 2021-11-06:
I've moved some things around, hope this is acceptible. 

[image]

brooksware2000, 2021-11-08:
I think you’re spot on with what I was thinking.  Agree on the load button
as well…should open a quick “Browse for file” option.

Thanks for working this in.

Curtis

closed 2022-05-19 by NotExpectedYet