OctoFarm/OctoFarm #301: Klipper/Octoprint compatibility

Automatica-Mat, 2021-02-17:
Hi,

I am trying to set up a small print farm and have recently converted a Corexy printer from Duet firmware to Klipper to try and integrate it with Octofarm as Klipper uses Octoprint. It seems there are some compatibility issues. The host state comes up as online however the is a No-API warning and the WebSocket is also showing offline. 
Klipper uses a file path as a printer path "/tmp/printer" but other than that I cant see why it shouldn't work.

Please let me know if there is something I could do to help figure out why Octofarm doesn't integrate with Kipper.

You could contact me on Skype: networkdevices.

Matt

davidzwa, 2021-02-17:
@Automatica-Mat we've discussed what could be going wrong and we think it has not much to do with Klipper.

Please verify (we need to be sure):
- OctoFarm version
- Relevant logs
- Screenshot of the printers page

About the error, please verify you've followed the OctoFarm guide on the README https://github.com/NotExpectedYet/OctoFarm#prerequisites:
- Created an API key with the first user on OctoPrint
- Enabled CORS on OctoPrint
- Used the first user API key within the API key field for adding a printer

Let us know if you run into further issues.

davidzwa, 2021-02-18:
For your information we've added https://github.com/NotExpectedYet/OctoFarm/issues/302

Automatica-Mat, 2021-02-18:
Hi Guys,

I had already added the API Key. I had done it twice to make sure. Please see the screen shots. Im running the Docker Monolithic version of Octofarm.

[cid:image005.png@01D70634.9C920400]

[cid:image006.jpg@01D70634.9C920400]
[cid:image003.png@01D70634.9C6DDA30]
[cid:image004.png@01D70634.9C6DDA30]

davidzwa, 2021-02-18:
Hmm it would be smarter to react on the Github issue instead of email. We cannot read these attachments.
https://github.com/NotExpectedYet/OctoFarm/issues/301

Automatica-Mat, 2021-02-18:
Right sorry I just read the text at the bottom of the email that said reply to the email was possible.
Ill post the screen shots on git hub

#301<https://apac01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FNotExpectedYet%2FOctoFarm%2Fissues%2F301&data=04%7C01%7C%7C3225d363eba54be696a408d8d3f27728%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637492385625578976%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=C1aQQx7zKe5KFXUj9WopM36NDuLTTHy3sBrFFTlQUUY%3D&reserved=0>

-

Automatica-Mat, 2021-02-18:
I had set the API as per the instructions. Here are some screen shots as requested.

[image]

[image]

[image]

[image]

davidzwa, 2021-02-18:
@Automatica-Mat the entered API key format is not correct.

Not sure why `matt/` is prefixed to it, but that should not be there.

NotExpectedYet, 2021-02-18:
> @Automatica-Mat the entered API key format is not correct.
> 
> Not sure why `matt/` is prefixed to it, but that should not be there. If we can improve on the instructions provided, also let me know.
> Awaiting your response whether the problem is fixed to close this issue.

OctoFarm doesn't look to have figured out which user he's using. It's undefined on the printer status screen and it should have correctly grabbed that. That's probably due to the entered API key with matt/ prefix which as you rightly said @davidzwa should not be there. OctoFarm doesn't add that at all. Nobody would be able to connect so I gather the user has added that in confusion maybe? Not sure. @Automatica-Mat could you confirm you typed that in please? or whether OctoFarm added it? I presume the former but I'd just like to double check.

NotExpectedYet, 2021-02-18:
If you could attach the OctoFarm-State logs please we can check what's exactly going on here.

Automatica-Mat, 2021-02-18:
I entered it that way as the words below the field where you enter the api key say "user/ api key"I will try again tomorrow evening.Thank you for your time.While I'm here could I suggest something. For my application I'm connecting this octofarm sever pc back to another network via VPN. This way I can easily access the farm stat's externally. The issue I'm facing is with the cameras. These won't work. It would be great if the server could proxy these streams. So everything is coming from the one ip. I'm sure there is a place to register these ideas I'll look tomorrow.Good night..Matt

NotExpectedYet, 2021-02-18:
There's work for that coming, the current client is all legacy code from when it's was a single html file. The streams should work fine over VPN though as long as they are within the VPN network. 

[image]
If you mean this part here I don't see another way to write that currenly as it's specifically refering to OctoPrint and a User or Application Key, not the "{usernam}/applicationkey" I will have a think because this is due to OctoPrint currently and it verging away from the Global API Key. We have some fixes to do but I'm stuck for time at the moment due to other commitments.

Automatica-Mat, 2021-02-20:
Hi yes it was just a misunderstanding by me about the key entry format.

I have deleted the printers and re added them. They are now putting this error on the screen when I try to enter the control page.

Something has gone wrong with loading the Printer Manager... Hard Failure, please submit as a bug on github: TypeError: Cannot read property 'baudrates' of undefined

Which log would you like me to send you?

Matt

davidzwa, 2021-02-20:
Please send us the contents of the log folder as a zip ;)

Automatica-Mat, 2021-02-20:
As mentioned Im running the Docker Monolithic Image. I when to the web page and downloaded each log and zipped it. I hope its enough.
[Uploading OctoFarm-HistoryCollection.zip…]()

davidzwa, 2021-02-20:
Im sorry the link you've just sent is not the zip.
Its also an option to drag the logs themselves into an issue, that way any ZIP issue is not gonna be a problem

davidzwa, 2021-02-20:
Btw, if you approach me on discord we can chat more freely!

https://discord.gg/vjabMUn
Username's the same

davidzwa, 2021-02-21:
Closing this issue. Please find us on discord for further questions.

closed 2021-02-21 by davidzwa

Automatica-Mat, 2021-02-21:
I sent you a message via discord. I punched a bunch of holes in my firewall temporarily to allow you access to the devices if you're interested in seeing whats going on.

Closed #301<https://apac01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FOctoFarm%2FOctoFarm%2Fissues%2F301&data=04%7C01%7C%7Cbf61f6aa5244413d7a2d08d8d642c256%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637494929507092292%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=cloq3qqjX4YHWtLZyOOnbuqhYCPL6JYgG1PxTFYF8dg%3D&reserved=0>.

-

davidzwa, 2021-02-21:
Please make sure to actually send it.