OctoFarm/OctoFarm #302: [NOTICE] OctoPrint API keys and CORS problems

davidzwa, 2021-02-17:
To all OctoFarm users having difficulty getting the OctoPrint API key to work, @NotExpectedYet and me want to share a bit of information and a plan we have in the near future.

# API keys
There are 3 different types of API keys in OctoPrint. 
- User API Key (useful). The user API key works only if the user has the groups **Admins** and **Operator**. This is the first user you make, so make sure to add the roles yourself for it to work for another user (if you must).
- Application Key (useful). The same as a User key but targeted for specific applications: like OctoFarm. 
Make sure your user has the groups **ADMINS** AND **OPERATOR** for your user - this is default for only the 1st user!
- Global API Key. We've had trouble using the Global API Key and it's **will not fully work + should not be used => dont use it**.

Quoting OctoPrint's Application Key page:
> The global API key should best not be used anymore. It's one single key that gives full admin access to your whole OctoPrint instance. **Instead of using it you should create individual Application Keys for your third party clients.** That way they get permissions matching the user account used for key creation and you can also revoke access to one app without having to change the keys for all other apps. It's also recommended to create a user account without admin access and use that for third party clients where possible.

TL;DR create an Application Key for the first OctoPrint account you made ❤️ 
This the way to get there:
- Settings wrench icon 
[image]
- Application Key (side navigation within OctoPrint Settings)
[image]
- Copy the generated key in the `Registered application keys` field which will appear after you click `Generate`
[image]

- (Optional) you can do this also in the User Settings, where the same Application Key are managed but only for the user itself, as well as also a user-specific API Key can be generated. Pretty confusing, yeah. Just use Application Key.

## How to check the user's  groups
- (Optional) make sure the user has Operator and Admins groups (which is default for the first user). 
> OctoPrint Settings (wrench icon) > Access Control > Edit the user you want (crayon icon)

[image]

# CORS setting
What happens If CORS is disabled?
> The OctoFarm website (a.k.a. the client) cannot call OctoPrint. The OctoFarm server can still do it, but that's not how its currently all setup.

The CORS setting being checked off results in the website of OctoFarm not being able to OctoPrint directly. Thus stuff wil fail.
TL;DR enable CORS please ❤️ 

# What our plans are
Right now if you enter a Global Key, we can use a small hack to check if it is indeed the global key and try to throw a warning. This is not something permanent and you should follow the instructions above.

About CORS, we will try and enable it for you, in a future release, given a correct API Key is provided to reach the OctoPrint client. This is not a guarantee on our side however.

Happy printing!
The devs

NotExpectedYet, 2021-02-17:
Legend thanks for that man!

NotExpectedYet, 2021-04-23:
No stale

davidzwa, 2021-04-23:
Ill play bot haha

lsls421, 2021-04-27:
Hi! Thanks for the effort of creating this guide, it is much appreciated. 

I would like to ask a short question: I have two admin user accounts in OctoPrint, and I actually want to use the second user's API for OctoFarm. Is it possible to tell OctoFarm to choose between multiple admin users instead of automatically choosing the first one?

NotExpectedYet, 2021-04-27:
> Hi! Thanks for the effort of creating this guide, it is much appreciated.
>
> I would like to ask a short question: I have two admin user accounts in OctoPrint, and I actually want to use the second [...]

Not currently no sadly. OctoFarm was written to save a bit of time in that regards for larger farms. It's worth adding a discussion for a feature request and we will look into adding it though. :)

lsls421, 2021-04-29:
> Not currently no sadly. OctoFarm was written to save a bit of time in that regards for larger farms. It's worth adding a discussion for a feature request and we will look into adding it though. :)

Thanks for the quick feedback. I can understand the reason for this. Feature request added in the discussion. :D

davidzwa, 2021-06-29:
@github-actions[bot] not stale my friend

closed 2021-07-11 by NotExpectedYet