OctoFarm/OctoFarm #309: [FEATURE IMPLMENTATION] Patreon: Add toggle option to check if spool is loaded on Printer before print.

NotExpectedYet, 2021-02-20:
Add a toggle option in settings for filamanent manager. 

Toggle will put a force check on the spool been loaded for the print.

3DCT-David, 2021-02-20:
Just for clarification, will this add a separate check to verify loaded or will this use the filament drop down selection (my preferred method)? I mainly want to verify the correct color is loaded and specified in the filament drop down as that will eventually set the temperature offset. Thanks!

davidzwa, 2021-02-20:
We're currently working super hard on the new UI, so whatever we'll do it will be much easier to do once the overhaul is complete :)

We'll keep you posted.

NotExpectedYet, 2021-02-20:
> Just for clarification, will this add a separate check to verify loaded or will this use the filament drop down selection (my preferred method)? I mainly want to verify the correct color is loaded and specified in the filament drop down as that will eventually set the temperature offset. Thanks!

Hmmm I thought the original request was just to check if it was loaded thanks for clarifying that. 

The colour verification is a little harder to do as it would depend on slicer, and current OctoFarm doesn't have any information of what's in the gcode file. 

Not saying it isn't possible it would just put this specific request back some.

NotExpectedYet, 2021-02-20:
Wait a minute... 

[image]

Does this still work? I haven't tried it in a while completely forgot it was there. If so what I had in mind is already a feature 😅. 

The colour check we can keep in mind for you. Would need slicer details and won't be available until the global file store is done though as that's the only way I'd get that colour choice if it's via slicer. If not maybe a name parse might work. 

If I've completely mis-understood let me know.

3DCT-David, 2021-02-21:
I'm not needing it to verify the "actual color" is loaded, just that when I click print it verifies that I selected "a filament/color" from the filament drop down menu. Basically to serve as a reminder: 1-Did I load the correct filament/color in the printer? and 2- Selecting the correct filament from the drop down will ensure that the corresponding temperature offset is used. I hope this makes sense.
From the softwares perspective I'd say that when I click "print" it checks the filament drop down and the default is null or what ever, it will than prompt me to select the filament. Once a selection (any selection) is made than it will allow me to print the job.

NotExpectedYet, 2021-02-21:
> I'm not needing it to verify the "actual color" is loaded, just that when I click print it verifies that I selected "a filament/color" from the filament drop down menu. Basically to serve as a [...]

Thanks for the further clarification. 
Could you confirm the current setting I linked works for you? If not this needs converting to an issue and adding to the current 1.1.11 plans. I won't get chance till maybe Monday afternoon so if you don't get chance no trouble at all. On it.

NotExpectedYet, 2021-02-27:
To confirm this doesn't work and I remember why now. 

It's when I added multi-tool support. 

Rather than holding 1.1.11 up anymore I will add this as a feature request for 1.1.12 as it's a bit of work.

NotExpectedYet, 2021-02-27:
Ignore me, it's file manager that doesn't check... the print button does. 

Will be fixed for 1.1.11

commit c9de0ed8 (2021-02-27): Updated
Fixed #309 - Just to clear
closed 2021-02-27 by NotExpectedYet