OctoFarm/OctoFarm #331: [QUESTION] Cannot start, what how to investigate

rsurgiewicz, 2021-02-26:
Hi Folks,

it seems that I'm doing something wrong getting error when trying to start octofarm:

```
0 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
1 info using npm@7.6.0
2 info using node@v14.16.0
3 timing config:load:defaults Completed in 4ms
4 timing config:load:file:/usr/lib/node_modules/npm/npmrc Completed in 4ms
5 timing config:load:builtin Completed in 4ms
6 timing config:load:cli Completed in 4ms
7 timing config:load:env Completed in 0ms
8 timing config:load:file:/var/lib/octofarm/.npmrc Completed in 0ms
9 timing config:load:project Completed in 2ms
10 timing config:load:file:/root/.npmrc Completed in 1ms
11 timing config:load:user Completed in 1ms
12 timing config:load:file:/usr/etc/npmrc Completed in 0ms
13 timing config:load:global Completed in 0ms
14 timing config:load:cafile Completed in 1ms
15 timing config:load:validate Completed in 0ms
16 timing config:load:setUserAgent Completed in 2ms
17 timing config:load:setEnvs Completed in 2ms
18 timing config:load Completed in 21ms
19 verbose npm-session d613d40a783e2b37
20 timing npm:load Completed in 40ms
21 timing command:run-script Completed in 11173ms
22 timing command:start Completed in 11179ms
23 verbose stack Error: command failed
23 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/index.js:64:27)
23 verbose stack     at ChildProcess.emit (events.js:315:20)
23 verbose stack     at maybeClose (internal/child_process.js:1048:16)
23 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
24 verbose pkgid octofarm@1.1.10
25 verbose cwd /var/lib/octofarm
26 verbose Linux 5.4.75-v7l+
27 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
28 verbose node v14.16.0
29 verbose npm  v7.6.0
30 error path /var/lib/octofarm
31 error command failed
32 error signal SIGTERM
33 error command sh -c pm2 flush && pm2 start app.js --name OctoFarm -e 'logs/pm2.error.log' --time
34 verbose exit 1
```

How to investigate deeper where's a problem?
Best!

davidzwa, 2021-02-26:
@rsurgiewicz we've shortly looked and found that you are running Node 14.

Please be so kind to install Node v12, as per instructions
https://github.com/OctoFarm/OctoFarm#prerequisites

We are working to fix the underlying database issues, but that needs some time!

rsurgiewicz, 2021-02-26:
Thanks @davidzwa 

however still no luck :/

```
0 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
1 info using npm@7.6.0
2 info using node@v12.21.0
3 timing config:load:defaults Completed in 4ms
4 timing config:load:file:/usr/lib/node_modules/npm/npmrc Completed in 4ms
5 timing config:load:builtin Completed in 4ms
6 timing config:load:cli Completed in 4ms
7 timing config:load:env Completed in 0ms
8 timing config:load:file:/var/lib/octofarm/.npmrc Completed in 1ms
9 timing config:load:project Completed in 2ms
10 timing config:load:file:/root/.npmrc Completed in 0ms
11 timing config:load:user Completed in 1ms
12 timing config:load:file:/usr/etc/npmrc Completed in 0ms
13 timing config:load:global Completed in 0ms
14 timing config:load:cafile Completed in 1ms
15 timing config:load:validate Completed in 0ms
16 timing config:load:setUserAgent Completed in 1ms
17 timing config:load:setEnvs Completed in 2ms
18 timing config:load Completed in 20ms
19 verbose npm-session 753fd50ccfcba27a
20 timing npm:load Completed in 40ms
21 timing command:run-script Completed in 10567ms
22 timing command:start Completed in 10573ms
23 verbose stack Error: command failed
23 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/index.js:64:27)
23 verbose stack     at ChildProcess.emit (events.js:314:20)
23 verbose stack     at maybeClose (internal/child_process.js:1022:16)
23 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
24 verbose pkgid octofarm@1.1.10
25 verbose cwd /var/lib/octofarm
26 verbose Linux 5.4.75-v7l+
27 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
28 verbose node v12.21.0
29 verbose npm  v7.6.0
30 error path /var/lib/octofarm
31 error command failed
32 error signal SIGTERM
33 error command sh -c pm2 flush && pm2 start app.js --name OctoFarm -e 'logs/pm2.error.log' --time
34 verbose exit 
```

NotExpectedYet, 2021-02-26:
Where did you clone the OctoFarm folder too? 

It's complaining about /var/lib/octofarm and node won't run as root so it would be dependent on your systems user permissions. 

Its recommended to just install it to your home folder for simplicity sake as we won't know every configuration a user has. 

Could you detail a little more about your system please?

NotExpectedYet, 2021-02-26:
I apologise as well I didn't see that before.

rsurgiewicz, 2021-02-26:
Thanks a lot,
Actually I'm trying to build my docker image for armv7 for rpi4 because the public one didn't work for me.

My Dockerfile:
```
FROM debian as gitcloner
RUN apt-get update && apt-get install -y git

FROM gitcloner as builder
RUN apt-get update && apt-get install -y curl build-essential autoconf curl dirmngr apt-transport-https lsb-release ca-certificates
RUN curl -fsSL https://deb.nodesource.com/setup_12.x | bash -
RUN apt-get install -y nodejs
RUN npm install npm@latest -g
RUN npm install pm2@latest -g

FROM builder
RUN git clone https://github.com/NotExpectedYet/OctoFarm.git /var/lib/octofarm && cd /var/lib/octofarm && git checkout 1.1.10

#FROM node:13-alpine

ENV MONGOUSER root
ENV MONGOPASS rootpassword
ENV MONGOHOST mongodb

#COPY --from=builder /git /var/lib/octofarm
WORKDIR /var/lib/octofarm
RUN mkdir logs
RUN sed -i "s/192.168.1.5:27017\/octofarm/$MONGOUSER:$MONGOPASS@$MONGOHOST:27017\/admin/g" ./config/db.js
RUN ls && npm install
EXPOSE 4000

RUN rm -Rf /var/lib/octofarm/node_modules/mozjpeg

# Dev mode
#CMD ["npm", "run", "dev"]

# Production mode
CMD ["npm", "start" ]
```

and my docker-compose:
```
version: '3'

services:
  octofarm:
    image: octofarm-armv7
    build:
     dockerfile: Dockerfile
     context: .
    container_name: octofarm-armv7
    environment:
      MONGOHOST: octofarm-mongo
      MONGOUSER: farmer
      MONGOPASS: octoprint
    ports:
      - 4040:4000
    volumes:
      - ${PWD}/logs/additional/:/root/.npm/_logs/
    restart: unless-stopped

  octofarm-mongodb:
    image: webhippie/mongodb:latest
    restart: unless-stopped
    environment:
      MONGO_INITDB_ROOT_USERNAME: farmer
      MONGO_INITDB_ROOT_PASSWORD: octoprint
      MONGO_INITDB_DATABASE: octofarm
    volumes:
      - ${PWD}/mongodata:/data/db

```

I'll keep trying maybe use different user within dockerfile if above error is caused by root usage.

Thanks anyway!

davidzwa, 2021-02-26:
You mentioned that the rpi4 didnt work for you. Lets start there.

What goes wrong? What OS are you running on it?

rsurgiewicz, 2021-02-26:
I really do appreciate Your efforts! 
Thanks for quick reply even though it's Friday afternoon (at least in my timezone) :-)

I have a raspian 32bit and yes I do ack that "regular" mongo image will not run.
I found a mongo docker image tailored for arm that works just fine (image in my docker-compose.yaml)

Errors I was getting were:
`standard_init_linux.go:211: exec user process caused "exec format error"`

But I have a bunch of containers running there so it's not kinda generic docker/system error I guess but rather problem with particular image

rsurgiewicz, 2021-02-26:
Small update:
after running npm in the "DEV"  way I got to the point I can see it's connected to mongodb and I could see a small stub of GUI in web browser but then it restarts with same manner
```

0 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'dev' ]
1 info using npm@7.6.0
2 info using node@v12.21.0
3 timing config:load:defaults Completed in 4ms
4 timing config:load:file:/usr/lib/node_modules/npm/npmrc Completed in 3ms
5 timing config:load:builtin Completed in 3ms
6 timing config:load:cli Completed in 5ms
7 timing config:load:env Completed in 1ms
8 timing config:load:file:/var/lib/octofarm/.npmrc Completed in 1ms
9 timing config:load:project Completed in 2ms
10 timing config:load:file:/root/.npmrc Completed in 0ms
11 timing config:load:user Completed in 0ms
12 timing config:load:file:/usr/etc/npmrc Completed in 1ms
13 timing config:load:global Completed in 1ms
14 timing config:load:cafile Completed in 0ms
15 timing config:load:validate Completed in 1ms
16 timing config:load:setUserAgent Completed in 1ms
17 timing config:load:setEnvs Completed in 2ms
18 timing config:load Completed in 20ms
19 verbose npm-session 8847b5e3d0cab818
20 timing npm:load Completed in 39ms
21 timing command:run-script Completed in 41896ms
22 verbose stack Error: command failed
22 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/index.js:64:27)
22 verbose stack     at ChildProcess.emit (events.js:314:20)
22 verbose stack     at maybeClose (internal/child_process.js:1022:16)
22 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
23 verbose pkgid octofarm@1.1.10
24 verbose cwd /var/lib/octofarm
25 verbose Linux 5.4.75-v7l+
26 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "dev"
27 verbose node v12.21.0
28 verbose npm  v7.6.0
29 error path /var/lib/octofarm
30 error command failed
31 error signal SIGTERM
32 error command sh -c nodemon ./app.js
33 verbose exit 1
```

davidzwa, 2021-02-26:
Haha yeah Friday! "Says 'hold my beer'"

Did you consider Raspberry Pi OS Lite? It's 64 bit, so it will not have the MongoDB problem. For the rest, Im having a hard time even thinking of how to reproduce this nor a clue what's going on :/

rsurgiewicz, 2021-02-26:
FGIF :)

Nope, I have actually a running system there with some home automation stuff so I'm not able just to eject it.

The thing is I'm 100% sure mongodb 32bit image works fine so I'd rather consider problem is elsewhere.

But anyway thanks for trying

davidzwa, 2021-02-26:
Yeah I agree that Node or whatever OF is doing is a 32-bit problem.
I presume you've also found SO's like this
https://stackoverflow.com/questions/42494853/standard-init-linux-go178-exec-user-process-caused-exec-format-error

So it's 32 bit biting us in the ass here

rsurgiewicz, 2021-02-26:
When using "my image"

looking at the mongodb logs i get:

`
octofarm-armv7      | [nodemon] restarting due to changes...
octofarm-armv7      | [nodemon] starting `node ./app.js`
octofarm-mongodb    | 2021-02-26T16:57:33.402+0000 [initandlisten] connection accepted from 172.29.0.3:36248 #1 (1 connection now open)
octofarm-mongodb    | 2021-02-26T16:57:33.481+0000 [initandlisten] connection accepted from 172.29.0.3:36250 #2 (2 connections now open)
octofarm-mongodb    | 2021-02-26T16:57:33.492+0000 [conn2]  authenticate db: octofarm { authenticate: 1, user: "octofarm", nonce: "xxx", key: "xxx" }
octofarm-mongodb    | 2021-02-26T16:57:33.528+0000 [initandlisten] connection accepted from 172.29.0.3:36252 #3 (3 connections now open)
octofarm-mongodb    | 2021-02-26T16:57:33.528+0000 [initandlisten] connection accepted from 172.29.0.3:36254 #4 (4 connections now open)
octofarm-mongodb    | 2021-02-26T16:57:33.530+0000 [initandlisten] connection accepted from 172.29.0.3:36256 #5 (5 connections now open)
octofarm-mongodb    | 2021-02-26T16:57:33.578+0000 [conn3]  authenticate db: octofarm { authenticate: 1, user: "octofarm", nonce: "xxx", key: "xxx" }
octofarm-mongodb    | 2021-02-26T16:57:33.580+0000 [conn4]  authenticate db: octofarm { authenticate: 1, user: "octofarm", nonce: "xxx", key: "xxx" }
octofarm-mongodb    | 2021-02-26T16:57:33.582+0000 [conn5]  authenticate db: octofarm { authenticate: 1, user: "octofarm", nonce: "xxx", key: "xxx" }
octofarm-mongodb    | 2021-02-26T16:57:33.583+0000 [initandlisten] connection accepted from 172.29.0.3:36258 #6 (6 connections now open)
octofarm-mongodb    | 2021-02-26T16:57:33.583+0000 [initandlisten] connection accepted from 172.29.0.3:36260 #7 (7 connections now open)
octofarm-mongodb    | 2021-02-26T16:57:33.604+0000 [conn6]  authenticate db: octofarm { authenticate: 1, user: "octofarm", nonce: "xxx", key: "xxx" }
octofarm-mongodb    | 2021-02-26T16:57:33.605+0000 [conn7]  authenticate db: octofarm { authenticate: 1, user: "octofarm", nonce: "xxx", key: "xxx" }
`

rsurgiewicz, 2021-02-26:
Hooolly molly 
I think I have a breakthrough :)

[image]

davidzwa, 2021-02-26:
Btw our image runs this within the ENTRYPOINT script
`pm2 start app.js --name OctoFarm --no-daemon -o './logs/pm2.log' -e './logs/pm2.error.log' --time`

pm2 helps a lot in error messages.

davidzwa, 2021-02-26:
Do tell, what was the fix?

rsurgiewicz, 2021-02-26:
Honestly speaking no idea...
I've purged all previous build images and recreated,

but still working in "dev" environment

davidzwa, 2021-02-26:
Ah too bad, well I'd like to invite you to https://discord.gg/vjabMUn to share any future problems. That way the community can help you as well and you can possibly help others.
Closing!

closed 2021-02-26 by davidzwa