OctoFarm/OctoFarm #341: System page error OctoFarm 1.1.11

doucettom, 2021-03-01:
I have the System page crashing since I upgraded on Ubuntu Server 20.04.2 64bits.
Here is the error:
https://pastebin.com/m166WJe6

NotExpectedYet, 2021-03-01:
Hey man, can you fill out the issue request form please?

doucettom, 2021-03-01:
> Hey man, can you fill out the issue request form please?
Where is it ?

doucettom, 2021-03-01:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Client or Server side error** 
Which part of the serve does this effect? If you don't know/client side then please add screenshots with the browser console "F12" open.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Main Page'
2. Click on 'System (upper right corner)'
4. See error 
https://pastebin.com/m166WJe6
**Expected behavior**
Open the system information page
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - NodeJS Version
 - srv@octofarm:~$ npm version
{
  npm: '7.5.3',
  node: '13.12.0',
  v8: '7.9.317.25-node.30',
  uv: '1.35.0',
  zlib: '1.2.11',
  brotli: '1.0.7',
  ares: '1.16.0',
  modules: '79',
  nghttp2: '1.40.0',
  napi: '6',
  llhttp: '2.0.4',
  openssl: '1.1.1e',
  cldr: '36.1',
  icu: '66.1',
  tz: '2019c',
  unicode: '13.0'
}

 - OctoFarm Version [e.g. 1.5.5.7] 1.1.11
 - OctoPrint Version [e.g. 1.4.1] N/A
 - OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius] N/A

**Additional context**
Ubuntu Server 20.04.2 amd64
[image]

NotExpectedYet, 2021-03-01:
A recursion error same as file manager by looks. Cheers, I'll back on it around Thursday so will get a bug fix release sorted.

NotExpectedYet, 2021-03-04:
Huh this is a weird one that key should exist and be blank although could be more gracefully handled. 

Would you mind passing me your System logs as well please? 

That key deals with your hard drive size so I'm a bit perplexed as to why that isn't grabbing it. I couldn't reproduce on my system here so I'd just like to make sure nothing else is going awry in the background causing that to fail.

NotExpectedYet, 2021-03-04:
It should be working fine on node 13, but I wonder if it's a slight difference in those systems. If you get time you could try downgrading to 12 and see if it persists. I gather it might still so no worries if you don't want the hassil as I'll have a fix for it on development shortly.

closed 2021-03-04 by NotExpectedYet

NotExpectedYet, 2021-03-04:
I believe this should be fixed now. 

If you `pm2 stop OctoFarm' 
`git pull` 
`npm start` 

It should make sure you get the patched dependency I use for that. 

If not then please do reopen and I'll investigate further.

commit 654aec29 (2021-03-04): Development 1.1.12 hotfix (#347)

* linter fight!

* make systemInfo null, check for such on client to stop page failing if unable to generate

* uncomment cpuInfo to stop failing..

* update repo url.. wow!

* Fixed #341 - System information was failing to update due older package version

* updated systeminfo package version Fixed #341

* Fixed #337 - Make sure file manager not loading files...

* update to 1.1.12-hotfix

* remove unneeded console.log

* add button for service restart of OctoFarm

* use npm update over npm install --production for prestart script. Better as whilst it also in [... cut]