OctoFarm/OctoFarm #342: Quotes still causing issues for some Windows 10 systems in package.json

NotExpectedYet, 2021-03-01:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Client or Server side error** 
Which part of the serve does this effect? If you don't know/client side then please add screenshots with the browser console "F12" open.

**Describe the bug**
A clear and concise description of what the bug is.

Gather it still doesn't like this, tested on my own windows 10 and didn't reproduce for me. Should just need them removing but will have to test further

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13]
 - OctoFarm Version [e.g. 1.5.5.7]
 - OctoPrint Version [e.g. 1.4.1]
 - OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius]

**Additional context**
Add any other context about the problem here.

davidzwa, 2021-03-03:
The quotes issue with pm2 folder seems to be fixed by using the ecosystem config
https://pm2.keymetrics.io/docs/usage/application-declaration/#log-files

{
 "errorFile": "logs/pm2.error.log"
 ...
}

closed 2021-03-04 by NotExpectedYet

davidzwa, 2021-03-04:
The comment I made about ecosystem was completely wrong ofc.