OctoFarm/OctoFarm #346: Filament Manager sync just one way?

beef76, 2021-03-04:
**Describe the bug**
Not sure if it's the actual design, but by synchronizing the filament manager with one octoprint the content from Octoprint is synced into the octofarm manager. If a new spool is added within Octofarm, those information are not synced back to the Filament Manager of the connected Octoprint. Also changes of the used filament are not synced to other octoprint instances filament manager.

Would be nice to have the content always synced between Octofarm and ALL connected Octoprint instances both ways, that all are up to date and in case you run jobs from an octoprint directly, the correct status of the Spools are used by selecting the spool on the actual used Octoprint.

NotExpectedYet, 2021-03-04:
OctoFarm grabs from OctoPrint plugin when a print is completed. When you add a new spool in OctoFarm does indeed update back to OctoPrint because it has to use the API to create the new spool, you would see errors if this fails. The changes on a completed print are also updated because they are update on OctoPrint itself so it's nothing to do with OctoFarm by the sounds. 

If they are not syncing between OctoPrint instances then you haven't followed the requirements and setup the postgres database by the sounds, which is a requirement of the plugin not OctoFarm. If this is not the case and you have indeed set it up as instructed below then I will need to see your filament manager logs to understand what it going on. 

Did you read this note? 
[image]

Please remember OctoFarm supports the plugin, not the other way around here. 

The only thing I don't currently track is if you add a spool on OctoPrint, that requires a re-sync.

beef76, 2021-03-04:
Hey James
Thanks for your quick reply! Much appreciated!
First of all i want to make clear that my issue reported above was not meant as any kind of complaint or so :-) don't get me wrong.
Thats also why i mentioned that this is maybe the design.

I have filament manager running on two octoprint instances, both on seperate Pi's. Both are configured with Octofarm and all is working well. The second one was just added lately.
I just recognized that by adding a new Spool via Octofarm and doing a manual resync of the filament manager, this spool was not synced back to the octoprint clients.
also the usage of a spool of one of the two Octoprints was synced to the filament manager on Octofarm, but not back to the other Octoprint filament manager data.
Both are setup as default and no external database connected or so.

If i add a spool to one of the Octoprint Filament managers, this is getting synced to the Octofarm one, but seems not the other way round.

Maybe the issue is sitting in front of the PC or something else weird is going on. If you say it should work both ways no matter where i setup a new spool, maybe you have any other hints or tips what i could check, please let me know.

Cheers
Bjoern

NotExpectedYet, 2021-03-04:
> Hey James
> Thanks for your quick reply! Much appreciated!
> First of all i want to make clear that my issue reported above was not meant as any kind of complaint or so :-) don't get me wrong.
> Thats [...]

I'm sorry if my reply came across like that, I'm tired and it wasn't intentional I was just trying to be clear. 

Can you confirm you have setup the postgres database on your OctoPrint plugin manager? 

Polling OctoPrint for the changes all the time is not something I really want to do which is what would be required to get the two way sync to stay in sync. I can't see anything in the websocket connection from filament manager so OctoFarm will need to be told that the spool has been updated on OctoPrint. The only other option is using OctoFarm for your spool management but that does require the postgres database installing on all of your instances. I don't support the plugin without that currently.

beef76, 2021-03-04:
no worries man, all good :-) 
But just to confirm, if i add a new spool on Octofarm, not on one of the Octoprints, it should be synced to both Octoprint instances if i do a manual sync, right? Thats actually not the case unfortunately.
If i'm not mistaken completely postgress is installed on both.
But for sure with two octoprint instances now i will primarly use octofarm to run jobs and select the spools ect.
With just one Octoprint before it did not really matter, but time is changing and more printers will be added now.

Stay safe
Bjoern

NotExpectedYet, 2021-03-04:
> no worries man, all good :-)
> But just to confirm, if i add a new spool on Octofarm, not on one of the Octoprints, it should be synced to both Octoprint instances if i do a manual sync, right? Thats [...]

I want to remove that damn postgres database dependency but is a ways off yet haha. 

If your filament manager database is setup and working then yeah totally. It will make an API call, then the instance will update the database but this does require at least one of those instances to be online. ie Octoprint alive. The database would then be read by your other instance. If it's updating on one but not the other I'd say it was OctoPrint and the database if it doesn't appear on any then it's me and OctoFarm aha.  

If you pass me the filament manager logs I can investigate and make sure it's not a bug on OctoFarm at least. If it is OctoPrint I can check your logs and advise as well. 

Jobs your fine starting from where ever you prefer  as the websockets will always keep those in sync. The file manager and filament manager currently just OctoFarm. 

Well I hope you a long and successful printing venture my man :) 

Oh I don't leave the house anyway don't worry 😂 but you too!

beef76, 2021-03-04:
ok man, i'm NOT using posgres as initially thought and actually not able to manage to get it running. 
Will investigate further and try my best to get it running on both Octoprint instances and try again. And yes i'm doing a backup first of all my spools :-) 

Thanks for all your help so far. Let me try further and keep this issue open here until i report back succes, okay?

thanks again for your kind support, it's so much appreciated!
Bjoern

NotExpectedYet, 2021-03-05:
> ok man, i'm NOT using posgres as initially thought and actually not able to manage to get it running.
> Will investigate further and try my best to get it running on both Octoprint instances and try [...]

I believe they carry over but don't quote me on that, best to be safe aha. 

No trouble, keep it open. 

If you let me know what your running I will be able to advise. I need to sort instructions for the new website anyway so will help me base those. 

Your most welcome too :)

beef76, 2021-03-05:
ok, got postgres running on both octoprint now, each has it's own, not one running the database and the other picking the data from the other, not sure if both will run 24/7.
Sync to octofarm worked well so far. But still the same if i add a new Spool on OCTOFARM :-(
If i add a new on one of the Octoprints, this is getting synced through the others without issues.
If this is the case, no problem to handle in the future and always add new spools on one of the Octoprints.

Weird thing i already faced before adjusting the Octoprints to use postrgesql is, that on octofarm just 5 spools have all information listed in the overview, the rest is not showing filament type ect. therefor also the filter is not working properly see screenshot. Any ideas? ;-)

[image]

NotExpectedYet, 2021-03-05:
> ok, got postgres running on both octoprint now, each has it's own, not one running the database and the other picking the data from the other, not sure if both will run 24/7.

As long as your database is on a Pi that's always on you'll be fine. You could install the PostGres database onto whatever you've got OctoFarm on to mitigate this as OctoFarm should be on an always on device. 

> Sync to octofarm worked well so far. But still the same if i add a new Spool on OCTOFARM :-(

OctoFarm will find the first online device to push the information to the database. Could you fire me through your OctoFarm-FilamentManager.log should be able to see if it's failing here. 

> If i add a new on one of the Octoprints, this is getting synced through the others without issues.
> If this is the case, no problem to handle in the future and always add new spools on one of the Octoprints.
> 

If you add on OctoFarm it should be syncing to them all. 

> Weird thing i already faced before adjusting the Octoprints to use postrgesql is, that on octofarm just 5 spools have all information listed in the overview, the rest is not showing filament type ect. therefor also the filter is not working properly see screenshot. Any ideas? ;-)
> 
> [image]

Ahh that's a bug I'd forgotten about and needs looking at. Basically if you open up the spools database, use the dropdown on that and set it to "All" that should display on the overview information. I should ignore that and pull the full list, but it seems to be conflicting somewhere. 

I've opened: https://github.com/OctoFarm/OctoFarm/issues/353 if you'd like to track it's status and added it to the 1.1.13 plans.

NotExpectedYet, 2021-03-05:
If that list still doesn't generate Material / Manufacturer then it's the filter on the profiles sorry. If you set that to All it will show :)

beef76, 2021-03-05:
> If that list still doesn't generate Material / Manufacturer then it's the filter on the profiles sorry. If you set that to All it will show :)

nothing to filter in the general overview unfortunately, it's the main page if you click on filament in octofarm where you also see the usage ect.

if you click on spool manager within the filament view it's showing all spools and i could filter by type or material. but i have the feeling something is wrong with my spools data in general...in this overview the manufacturer and type does not match the name/color...looks like something got really messed up and mixed here

[image]

NotExpectedYet, 2021-03-05:
> > If that list still doesn't generate Material / Manufacturer then it's the filter on the profiles sorry. If you set that to All it will show :)
> 
Sorry, it's on the profile pop-up, and spool pop-up with the buttons at the top. 

> nothing to filter in the general overview unfortunately, it's the main page if you click on filament in octofarm where you also see the usage ect.
>
> if you click on spool manager within the filament [...]

Hmm, I'll certainly take a look at that I didn't notice the mix up on my personal one.

beef76, 2021-03-05:
within spool manager view popup, all looks good, also no mix ups of manufacturer and material type ect and filter is working fine. 
Just in the general overview where usage by day is shown, the list of filaments shown below is mixed up like in the screenshot i shared earlier. It's just listing details for the first 5 spools and they are mixed up with Type and Manufacturer. 
As you can see in this new screenshot now from the Spool manager popup "aircraft grey" is from FormFutura and PLA, not PETG from "Das Filament" as shown in the overview table :-)

[image]

NotExpectedYet, 2021-03-24:
Closing this in favour of #353 #358 

Will be added in a PR to development shortly to fix this.

PR #390 (2021-03-24): Bugfix/filament manager cleanup
Major clean up of filament manager. 

To Do: 
   - [x] Fixed #346 
   - [x]  Fixed #353 
   - [x] Clean up of the ejs template loading values not disabled
   - [x] Enable values for editing, then disable again on save
   - [x] Fix issue with overview table not updating fields when filter applied to spool manager
   - [x] Fix issue with overview table not displaying correct values
   - [x] Added conditional turnary operator to big please @davidzwa "?." 
   - [x] 20 Squats to appease the printing gods 
   - [x] 10 pushups to appease big D
   - [x] Fix adding a spo [... cut]
closed 2021-03-24 by NotExpectedYet