OctoFarm/OctoFarm #351: BG image not ignored, make it a fallback image within assets

davidzwa, 2021-03-05:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Client or Server side error** 
Client

**Describe the bug**
Image can be changed, but then it becomes a git change as the file is still tracked.
Change image file and you'll see.

**Solution**
Make the image a fallback image with some smart index + EJS rendering logic.

NotExpectedYet, 2021-03-05:
One for 1.1.13 :) 

Will investigate a graceful way as currently is sat in CSS hard coded I believe.

davidzwa, 2021-03-05:
Point the URL to a route and figure it out in ExpressJS :)

closed 2021-04-24 by NotExpectedYet