OctoFarm/OctoFarm #363: Github Action to build client automatically on dev changes

NotExpectedYet, 2021-03-13:
Add in client to be built automatically on changes pushed to development branch.

Gather this would work, as when we PR to master all will be ready. 

Maybe have a 3rd branch called staging and automatically PR those to development

davidzwa, 2021-03-17:
Not very hard, just use the release workflow Ive used on the OctoFarm installer repo. The implication is that we should not release ourselves, but instead push git tags to trigger the workflow. You ok with adding that here as well? 

- The release will then have assets attached which people can download.
- We can optionally mark automatic releases as prerelease by default and then manually mark them as non-prerelease? The OctoFarm Installer filters on this automagically. Gives us a safety net to fallback to in case the release is problematic.

NotExpectedYet, 2021-03-19:
The git tags sounds fine to me, we can do pre-release git tags. Would we need the staging branch? I feel we would but maybe I am mis-understanding the flow somewhere. 

I gather the assets will be packaged inside anyway right? So they'd come with the full .zip from the release or the deb/OctoFarm installer build. 

Yeah that sounds epic, we should be able to keep the development branch for bleeding edge features and testing as well then. 

Very nifty :D

NotExpectedYet, 2021-03-19:
That actually gives me a good hook for the roadmap for the website too so liking the sounds of this even more.

davidzwa, 2021-03-19:
Some context to how a release looks:
https://github.com/OctoFarm/OctoFarm-Installer/releases/tag/0.1.2-prerelease
So we can download an asset, which is a compiled/prepped version of OctoFarm (zipped, targz or binary)

davidzwa, 2021-03-21:
@NotExpectedYet I will start work on this using gulp and add commented out webpack command to switch to later.
Will end up being a github action for pushed tag causing prereleases.

NotExpectedYet, 2021-03-23:
> > The git tags sounds fine to me, we can do pre-release git tags. Would we need the staging branch? I feel we would but maybe I am mis-understanding the flow somewhere.
> 
> Personally, my intuition tells me to link release tags to a single branch only (master) and to alpha (in future). We can mark unstable or nightly tags on dev, but only if we think it through so we dont end up with a shitload of releases.
> With alpha we already have split up our maintenance work, so separate releases make sense to me. The tags we push would have the `prerelease: true` setting when we push them in the Github Action. That way we can choose to intervene if the automation is not going right.
> 
> > I gather this could be also an apt time to bring in the automatics changelog as well?
> 
> About automatic changelog, whatever feels natural but I think the release action does a pretty decent job already. Up to you!
> 
> Some context to how a release looks:
> https://github.com/OctoFarm/OctoFarm-Installer/releases/tag/0.1.2-prerelease
> So we can download a specific asset within the release, which could be a compiled/built/prepped version of OctoFarm (zipped, targz or binary).

Yeah I think that been on the release is absolutely fine, nobody reads the change log tbh. I grab most of my website stuff from the release section anyway for a full release so I think the specific file is pretty pointless keeping up to date. 

[image]

Can this section be enhanced further? like user who did the PR/Commit? and sectioned off for Fixed, Changed, Added, Security?

davidzwa, 2021-03-23:
We can fully adjust it to whatever we want! We can also fetch a file from the repo f.e. and parse that.

NotExpectedYet, 2021-03-23:
Sexy!

davidzwa, 2021-03-23:
So concerning release notes, what has your preference? How shall we tackle it? (Like what source of info to use, we can agree to push certain files to the repo or we can use the github PR's closed since... options options)

Btw, I removed the test tag. Will keep the prerelease for testing purposes and delete it later.

NotExpectedYet, 2021-03-23:
Happy for more input but I personally believe we should use the PR merged, would it be? or closed if I'm mis-understanding that, once those have hit master. 

Development I don't think there should be an automated pre-release unless it's something we specifically want testing otherwise they'd be a pre-release every push. Those would be what I'd consider RC releases, so release candidates. 

We need a PR format so pre-appended, or tagged with "Fixed", "Changed", "Added", "Deprecated", "Security". 

Those should create a similar thing to the changelog is now so organised via headings. With the github username of the person who made the original PR work appended to the actual action but only in the Releases place. 

If this can create a running change file for the changelog.md as well even better. Any development pushes could just update the [UnReleased] tag or [Pre-Release] Whatever is more fitting, I prefer UnReleased currently. 

Extra bonus points if it can say Big Thanks to the following users at the bottom, and include a count of their pushup-# points XD. Or even if they pull through next to the actual commit name. Don't waste time on this like was more just a jokey/want.

davidzwa, 2021-04-24:
Id like to have this closed. Did the test earlier for gulp to be automated, didnt like the result. It broke some PR checks.

closed 2021-07-11 by NotExpectedYet