OctoFarm/OctoFarm #364: Incorrect Print Progress

tophattwaffle, 2021-03-14:
Printers will report incorrect status, for instance 100% when really at 70%.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a local print on octoprint.

[image]
The real progress of this print is ~75%

What Octoprint shows:
[image]

**Versions:**
 - Docker Monolithic-Latest
 - OctoFarm Version 1.1.12
 - OctoPrint Version: OctoPrint 1.5.3
 - OctoPi: 0.17.0

NotExpectedYet, 2021-03-18:
Did some testing on this and it seems to get slowly out of sync somewhere.

NotExpectedYet, 2021-03-24:
Is this still an issue @tophattwaffle I've done some more investigating and all I could find was a discrepancy with the estimated print time not aligning with OctoPrints but that isn't what I use to generate the progress. 

It seems my progress was more inline with the actual percentage too as OctoPrint rounds to the nearest single digit which i've not replicated on a PR i'm about to make. 

Neither of those would cause the massive discrepancy you are seeing though so I'm a little confused at the moment.

NotExpectedYet, 2021-03-24:
I have PR'd the request. When this hits development I'd love to get your feed back to see if it actually resolved the issue.

tophattwaffle, 2021-03-24:
what steps do I need to take to see if this is resolved? I'm running 1.1.12
on docker monolithic, so I'm assuming I'll have to wait for the next build
of that.

davidzwa, 2021-03-24:
> what steps do I need to take to see if this is resolved? I'm running 1.1.12 on docker monolithic, so I'm assuming I'll have to wait for the next build of that.
> […](#)

Once the pull-request is merged by @NotExpectedYet it will start building on `monolithic-dev`. This image would then have the latest work. If you have a moment, we can notify you once it is ready

NotExpectedYet, 2021-03-24:
That's now complete @tophattwaffle should be built on the latest development build.

NotExpectedYet, 2021-04-01:
Fingers Crossed! XD

tophattwaffle, 2021-04-01:
Looks good now! You guys rock.

NotExpectedYet, 2021-04-02:
I'm just gonna leave this open incase anyone else reports it and can see it's fixed.

commit 57ef1513 (2021-03-24): Round completion server side, remove unneeded rounding client side.
Fixed #364
commit 10456cc6 (2021-03-24): Round completion server side, remove unneeded rounding client side.
Fixed #364
commit 80dd2a4c (2021-03-24): Bugfix/incorrect print progress (#392)

* Fix estimated print time not updating with job. OctoPrint doesn't update the initial job.estimatePrintTime key but instead calculates from job.printTime + job.printTimeLeft

* Round completion server side, remove unneeded rounding client side.
Fixed #364

* Rebuild assets with gulp

* floor the progress.completion value rather than rounding.
commit 86791b12 (2021-04-24): Bugfix/incorrect print progress (#392)

* Fix estimated print time not updating with job. OctoPrint doesn't update the initial job.estimatePrintTime key but instead calculates from job.printTime + job.printTimeLeft

* Round completion server side, remove unneeded rounding client side.
Fixed #364

* Rebuild assets with gulp

* floor the progress.completion value rather than rounding.
closed 2021-04-24 by NotExpectedYet