OctoFarm/OctoFarm #374: Couple server port to .env and ditch serverSettings prop

davidzwa, 2021-03-17:
Server

- Migration script to check if serverSettings contains property for port, save to .env file (**non-docker**) as OCTOFARM_PORT and delete it from database.
- Let the UI not save it to database, but adjust the .env file instead or in case of docker disable the button with the instructions to change the env variable OCTOFARM_PORT and the container inner port instead (default:4000)
- Test the API port with smoke test
- Test a custom env variable API port with smoke test

NotExpectedYet, 2021-03-17:
Ok ignore my remove the UI option ahah

NotExpectedYet, 2021-03-19:
I believe I save the port regardless btw so might just have to check if it differs from port 4000 than whether it exists. This sounds grand though and keeps the current feature set in the UI in place.

davidzwa, 2021-04-22:
Removed this from the PR as I dont think it is fixed by it. The PORT still needs to be cleaned up in mongodb settings

davidzwa, 2021-04-27:
Has some value still

NotExpectedYet, 2021-04-27:
> Has some value still for 1.1.14

Oh sorry aha no worries

closed 2021-09-04 by github-actions[bot]