OctoFarm/OctoFarm #378: Settings Memory pie chart svg error

davidzwa, 2021-03-17:
**Client**
[image]
SVG errors or at least some NaN sizing properties.

Not a big problem, low prio as I see it.

davidzwa, 2021-03-17:
[image]
OctoRAM NaN probably

NotExpectedYet, 2021-03-18:
Deffo the NaN hmm weird I sorted all those. What system are you on? Windows?

davidzwa, 2021-03-18:
Jep windhose always

NotExpectedYet, 2021-03-18:
Consequence of webpack? Don't see how like that's that's a server side response thought I'd check.

davidzwa, 2021-03-18:
I think it's a division by zero + some missing data

NotExpectedYet, 2021-03-19:
Yeah deffo missing data, I will have to create a fall back and test on my windows box to see what's missing/changed as the value for total RAM should certainly be available. That system check isn't the most graceful thing in the world and needs a good cleanup/refactor itself.

davidzwa, 2021-04-08:
`systemInfo.sysProcess.mem` is not set. That's

We have the following properties in that subsection:
```
pmem = 2.009567407507439
memVsz = 326780
memRss = 335716
```
I went for `memRss` as it is described to represent used memory. Oh btw, I left `mem` as fallback, so your situation wont break.

davidzwa, 2021-04-08:
Fixed in PR #417 
[image]

davidzwa, 2021-04-24:
Different error now, needs new issue

closed 2021-04-24 by davidzwa