OctoFarm/OctoFarm #380: Update the API endpoint for the database URL

NotExpectedYet, 2021-03-19:
Not a priority but the database URL API endpoint the client uses currently pulls from the incorrect key as it wasn't grabbing from the new .env file method. 

Quick fix. 

### To better help the developer understand your 
issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Client or Server side error** 
Which part of the serve does this effect? If you don't know/client side then please add screenshots with the browser console "F12" open.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13]
 - OctoFarm Version [e.g. 1.5.5.7]
 - OctoPrint Version [e.g. 1.4.1]
 - OctoPrint Plugins [e.g. bed visualiser, octoklipper, printtimegenius]

**Additional context**
Add any other context about the problem here.

davidzwa, 2021-04-17:
So it is using the dotenv config right now. 

- Are we sure we want to expose username/password like this?
- Is this system page admin only?
- Do we want to make it adjustable there? It could cause a booting problem, but then we do give some control back to users

NotExpectedYet, 2021-04-17:
What do you mean username and password? 

The majority is admin user, I think the entire bottom panel doesn't show to a user permission. 

I'm not sure if it should be in the UI it was only there originally to help people change it but now we have environment it might not be needed what do you think?

davidzwa, 2021-04-17:
> What do you mean username and password?

I think having an authenticated database is super important. I dont understand why MongoDB goes without by default.

> The majority is admin user, I think the entire bottom panel doesn't show to a user permission.

Good!

> I'm not sure if it should be in the UI it was only there originally to help people change it but now we have environment it might not be needed what do you think?

Will answer offline

NotExpectedYet, 2021-04-17:
> > What do you mean username and password?
> 
> I think having an authenticated database is super important. I dont understand why MongoDB goes without by default.
> 
Up to the user, I run without Auth because it's all contained in its own segregated network. No idea on why mongo does it that way like.

closed 2021-04-24 by NotExpectedYet