OctoFarm/OctoFarm #386: Make sure update notification can only been seen by admin user.

NotExpectedYet, 2021-03-21:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Client or Server side error** 
Server

**Describe the bug**
Currently the update check that's run server side can be seen by any user using OctoFarm. This should at least be disabled for anyone who is not an admin. 

**Additional context**
Extra bonus points 20 pushups

davidzwa, 2021-04-17:
@NotExpectedYet this has to go into 1.1.13 I guess or is it less prio?

NotExpectedYet, 2021-04-17:
I'd say so, no idea on users / admin setup across my user base so rather be safe.

closed 2021-04-24 by NotExpectedYet