OctoFarm/OctoFarm #387: OctoFarm: Database Warning

TomKamin1, 2021-03-21:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Client or Server side error** 
Which part of the serve does this effect? If you don't know/client side then please add screenshots with the browser console "F12" open.
Server side error.

**Describe the bug**
A clear and concise description of what the bug is.
I installed OctoFarm Monolythic on my UNRAID server and it worked flawlessly, but after a day (with no server restarts or changes mind you) it randomly started displaying a "Database warning" error whenever i try to go into the UI. I tried reinstalling it, wiping all its data manually, nothing helpled.

**To Reproduce**
Steps to reproduce the behavior:
I do not know exactly, but ive not been able to even fix it.

**Expected behavior**
A clear and concise description of what you expected to happen.
I expected it to run.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Versions (please complete the following information):**
I do not have a clue at what is the newest version at the moment. I just dont know my way around unraid enough to figure it out, but its "newest" from like 3 days ago. 

**Additional context**
Add any other context about the problem here.

NotExpectedYet, 2021-03-23:
Hmmm weird one. Have you got any logs we could look at? They would be in your logs folder in appdata/OctoFarm/logs. 

Could you also let us know which tag your using? I gather it's latest but just wanted to confirm.

davidzwa, 2021-04-09:
@TomKamin1 we'd need more info on your side! Please provide the OctoFarm-Server logs in the logs folder as there the MongoDB error will be found.
Otherwise we must postpone any possible fix to another release.

TomKamin1, 2021-04-09:
Hey. Latest. My server is currently offline on maintanance so i don't have
any logs sadly. Once it's up if the problem persists I'll find the logs for
ya.

NotExpectedYet, 2021-04-19:
It should be in your app/appdata folder for OctoFarm. If you set the defaults then /mnt/user/appdata/OctoFarm/logs

TomKamin1, 2021-04-20:
So looking into that exact folder, there is nothing. Seriously, there's not
a single file in logs...

TomKamin1, 2021-04-20:
I did find this. Its not in a file, but when i click on the "octofarm app"
in unraid UI i can get this log out.
2021-04-20T10:38:24: PM2 log: [--no-daemon] Continue to stream logs
2021-04-20T10:38:24: PM2 log: [--no-daemon] Exit on target PM2 exit pid=11
10:38:35 0|OctoFarm | 2021-04-20T10:38:35: Tue, 20 Apr 2021 08:38:35 GMT |
ERROR | OctoFarm-InformationCleaning.log | Failed to generate clean
history... | data:"Operation `histories.find()` buffering timed out after
10000ms" |
10:38:35 0|OctoFarm | 2021-04-20T10:38:35: MongooseError: Operation
`profiles.find()` buffering timed out after 10000ms
10:38:35 0|OctoFarm | 2021-04-20T10:38:35: at Timeout.<anonymous>
(/app/node_modules/mongoose/lib/drivers/node-mongodb-native/collection.js:185:20)
10:38:35 0|OctoFarm | 2021-04-20T10:38:35: at listOnTimeout
(internal/timers.js:531:17)
10:38:35 0|OctoFarm | 2021-04-20T10:38:35: at processTimers
(internal/timers.js:475:7)
10:38:55 0|OctoFarm | 2021-04-20T10:38:55: Tue, 20 Apr 2021 08:38:55 GMT |
ERROR | OctoFarm-Server.log | MongooseServerSelectionError: connect
ECONNREFUSED 127.0.0.1:27017 |
10:38:55 0|OctoFarm | 2021-04-20T10:38:55: You have database issues... web
interface loaded on: 4000

davidzwa, 2021-04-20:
@NotExpectedYet could it be that the 127.0..0.1 does not resolve to MongoDB in his monolithic container? That's the only thing I can think of.... or an architecture issue which causes Mongo not to start as a service?

@TomKamin1 we also have 2 other containers. If you'd like you can setup the normal OctoFarm container and a separate MongoDB container. That way you can see much clearer what is going wrong with MongoDB.

TomKamin1, 2021-04-20:
Yeah, im gonna try the separate mongoDB now. Ill let you know once i set it
up.

TomKamin1, 2021-04-20:
So even after making it manually, i get the same error in the UI. Here are some logs i manged to pull.
[pm2.log](https://github.com/OctoFarm/OctoFarm/files/6342154/pm2.log)
[pm2.error.log](https://github.com/OctoFarm/OctoFarm/files/6342155/pm2.error.log)
[OctoFarm-Server.log](https://github.com/OctoFarm/OctoFarm/files/6342156/OctoFarm-Server.log)
[OctoFarm-InformationCleaning.log](https://github.com/OctoFarm/OctoFarm/files/6342157/OctoFarm-InformationCleaning.log)

NotExpectedYet, 2021-04-20:
> @NotExpectedYet could it be that the 127.0..0.1 does not resolve to MongoDB in his monolithic container? That's the only thing I can think of.... or an architecture issue which causes Mongo not to start as a service?

Yeah it wouldn't resolve at all, just checked the entry point for it. It use to work with that IP though and weird how it just suddenly stopped working. It's a custom edit to the config/dB.js in the entry point and it's been that way for 11 months so bit confused at the moment. 

The docker internal network is 172.17.0.1. Though so technically shouldn't work, even with host mode or whatever it would never understand the local range.

davidzwa, 2021-04-20:
Monolithic is internal though, so 127.0.0.1 loopback is about internal networking for the container itself.

davidzwa, 2021-04-20:
> So even after making it manually, i get the same error in the UI. Here are some logs i manged to pull.
> [pm2.log](https://github.com/OctoFarm/OctoFarm/files/6342154/pm2.log) [...]

Please be clear what you mean with 'manually'. Assuming that you setup 2 containers: 1 for MongoDB and 1 for OctoFarm, did you add the MONGO=...url_here... environment variable to the container for OctoFarm?

NotExpectedYet, 2021-04-20:
> So even after making it manually, i get the same error in the UI. Here are some logs i manged to pull.
> [pm2.log](https://github.com/OctoFarm/OctoFarm/files/6342154/pm2.log) [...]

@davidzwa in pm2.error.logs that's the node error I use to get with node version issue I mentioned. All the dockers are on 13 nodejs. So confused here, at work will take a deeper look when I get chance. Certainly needs updating to 14.

TomKamin1, 2021-04-20:
> Please be clear what you mean with 'manually'. Assuming that you setup 2 containers: 1 for MongoDB and 1 for OctoFarm, did you add the MONGO=...url_here... environment variable to the container for [...]

I followed this guide, step by step. 
https://forums.unraid.net/topic/90385-guide-for-octofarm-3d-printer-octoprint-farm-software-for-unraid/

TomKamin1, 2021-04-20:
Guys, i managed to fix it. I installed a plugin called "CA Cleanup Appdata" and had it clean everything in my appdata folder. Now i installed monolythic and it has no database issue anymore. Ill see if it runs ok from now on ad doesent randomly break like before, and let you know if it does. If theres any info you need, let me know. Thanks !

NotExpectedYet, 2021-04-20:
> Monolithic is internal though, so 127.0.0.1 loopback is about internal networking for the container itself. Or am I misunderstanding your point maybe?

I'm probably mis-understanding docker tbh. I've not the foggiest if it uses internal loopback just going off my own compose issues I've had with 127.0.0.1.

NotExpectedYet, 2021-04-20:
> Guys, i managed to fix it. I installed a plugin called "CA Cleanup Appdata" and had it clean everything in my appdata folder. Now i installed monolythic and it has no database issue anymore. Ill see [...]

If you followed that guide it's old the paths it tells you to setup no longer exist. We have UnRaid images in the CA app store with everything pre-configured for you. That does explain why you had no logs originally though, can't remember with that guide and also why the issue might have occured as you'd have had no config/dB.js file when the container restarted.

closed 2021-04-24 by davidzwa