OctoFarm/OctoFarm #418: DateTime formatting and internationalization

davidzwa, 2021-04-07:
Related to #255 I considered introducing date-fns, or MomentJS, but either are not the newest. Luxon is a really nice library with super tight API. It's created by one of the devs of MomentJS without the downsides of MomentJS old architecture.

Site
https://moment.github.io/luxon/index.html

How I would provide argument for using it over vanilla dates, date-fns or MomentJS:
https://stackoverflow.com/questions/10645994/how-to-format-a-utc-date-as-a-yyyy-mm-dd-hhmmss-string-using-nodejs/10717081#10717081

NotExpectedYet, 2021-04-07:
> Related to #255 I considered introducing date-fns, or MomentJS, but either are not the newest. Luxon is a really nice library with super tight API. It's created by one of the devs of MomentJS [...]

I think some of the history issues the user was experiencing should be fixed with the changes I've made to the date time functions. I'd not used the locale variant of them. System side should be set by the system personally, and using those functions client side would follow that too. 

I'm not apposed to a library / user selection option though.

davidzwa, 2021-04-07:
Ah, ignore the issue I referred to. I misread. Just to clarify:

Im reviewing code like the following:
```
    let currentDate = new Date();
    let dateComplete = "";
    currentDate = currentDate.getTime();
    const futureDateString = new Date(
      currentDate + printTimeLeft * 1000
    ).toDateString();
    let futureTimeString = new Date(
      currentDate + printTimeLeft * 1000
    ).toTimeString();
    futureTimeString = futureTimeString.substring(0, 5);
    dateComplete = `${futureDateString}: ${futureTimeString}`;
```
Which can be replaced by a single line of code with Luxon.
https://moment.github.io/luxon/docs/manual/formatting.html#formatting-with-tokens--strings-for-cthulhu-

NotExpectedYet, 2021-04-08:
> Ah, ignore the issue I referred to. I misread. Just to clarify:
>
> Im reviewing code like the following:
>
> ```
>     let currentDate = new Date();
>     let dateComplete = "";
>     currentDate = [...]

Aye certainly much better code wise, I'll take a look at the module as it's relevant to the history re-write cheers!

closed 2021-06-16 by davidzwa