OctoFarm/OctoFarm #429: Memory SVG on System page incorrect for certain Unix systems

davidzwa, 2021-04-09:
Server side

When fixing #378 I already noticed weird scaling and missing child property `mem`. Now the error is gone, but the calculation for non `memRss` calculations is off (I didnt change anything here).
Needs to be determined where the problem lies and rectified for that situation. I expect somehow `mem` is a different unit than kB (f.e. MB's or bytes).

NotExpectedYet, 2021-04-17:
Gonna move to a later release. Needs some investigating and testing across multiple systems. I've never had anyone point out it was wrong before but the memory one is certainly off somehow on my unix system.

davidzwa, 2021-04-17:
Not that big of a deal. The page works so thats good

commit 75648ece (2021-04-27): Fixed #429 - Memory issue on system page. (#474)

* Fixed #429 - Memory issue on system page.

* removed .?

* Squash memory svg work

* Update sys info jsdoc
commit 1248be6b (2021-04-27): Fixed #429 - Memory issue on system page. (#474)

* Fixed #429 - Memory issue on system page.

* removed .?

* Squash memory svg work

* Update sys info jsdoc
closed 2021-04-27 by NotExpectedYet