OctoFarm/OctoFarm #434: Filament use is tracked but the spools won't update after printing

Gagonjh, 2021-04-14:
**Describe the bug**
After printing, the weight of the roll is not adjusted. The statistics show the consumption but not in the material overview. 
In addition, the filament overview only shows 34 filaments with all the information. Both errors also occur after the database has been emptied and refilled.
[OctoFarm-FilamentManager (1).log](https://github.com/OctoFarm/OctoFarm/files/6309825/OctoFarm-FilamentManager.1.log)
[OctoFarm-HistoryCollection (1).log](https://github.com/OctoFarm/OctoFarm/files/6309826/OctoFarm-HistoryCollection.1.log)

**Screenshots**
[image]
[image]
[image]
[image]

System
Raspberry Pi 4 2GB (Server and on one 3D printer)
Raspberry Pi 3 1GB
Klipper as Firmware

Versions (please complete the following information):
Ubuntu (20.04.2 LTS)
OctoFarm (1.1.12)
Octopi (0.18.0)
Octoprint (1.5.3)
Bed Visualizer (1.0.1)
Marlin GCode Documentation (0.12.0)
OctoKlipper (0.3.7)
Preheat Button (0.7.2)
PrintTimeGenius Plugin (2.2.7)
PrusaSlicer Thumbnails (0.1.4)
Themeify (1.2.2)

NotExpectedYet, 2021-04-17:
That is only available when using Filament Manager Plugin. There is no automatic feature without that yet. If you'd like that it needs to be become a Discussion because it comes with caveats, especially around failed prints. 

I will close this as it's not a bug.

closed 2021-04-17 by NotExpectedYet

NotExpectedYet, 2021-04-17:
For the other issue please see 

Issues: #353 #358

Both should be fixed on the development branch and will be released with 1.1.13.