OctoFarm/OctoFarm #439: Problem with printer data (temp, time)

3DRemake, 2021-04-19:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Client or Server side error** 
Client error
Uncaught (in promise) TypeError: Cannot set property 'innerHTML' of null
    at panelViewRunner.min.js:6504
    at u (serverAliveCheck.min.js:6554)
    at Generator._invoke (serverAliveCheck.min.js:6784)
    at Generator.next (serverAliveCheck.min.js:6609)
    at g (panelViewRunner.min.js:5864)
    at o (panelViewRunner.min.js:5866)
(anonymous) @ panelViewRunner.min.js:6504
u @ serverAliveCheck.min.js:6554
(anonymous) @ serverAliveCheck.min.js:6784
(anonymous) @ serverAliveCheck.min.js:6609
g @ panelViewRunner.min.js:5864
o @ panelViewRunner.min.js:5866
Promise.then (async)
g @ panelViewRunner.min.js:5864
o @ panelViewRunner.min.js:5866
(anonymous) @ panelViewRunner.min.js:5866
(anonymous) @ panelViewRunner.min.js:5866
(anonymous) @ panelViewRunner.min.js:5954

**Describe the bug**
If you don't connect the 1st printer of the list you can't see data for the second printer in panel tab or in camera tab (camera is ok, but data like time and temp are not ok)

**To Reproduce**
Steps to reproduce the behavior:
1. Have at least 2 printer registered, shutdown the octopi of the first printer in the list
2. Lanch a print on the second printer
3. Go to 'Panel' tab
4. See data abour temp and time are not refreshed

**Expected behavior**
Data should be refresh as they are if the first printer is connected

**Screenshots**
Screenshots of my second printer (First printer offline)
https://media.discordapp.net/attachments/709293261646200842/831923458534932530/unknown.png

Screenshots of my second printer (First printer online)
https://media.discordapp.net/attachments/709293261646200842/831926710332817438/unknown.png

**Versions (please complete the following information):**
 - NodeJS Version [12.22.1]
 - OctoFarm Version [1.1.12]
 - OctoPrint Version [1.5.3]
 - OctoPrint Plugins [N/A]

**Additional context**
N/A.

davidzwa, 2021-04-19:
Thanks to @3DRemake the place in code was found. It most probably resides here:
[image]

tool element is not found

3DRemake, 2021-04-19:
> Thanks to @3DRemake the place in code was found. It most probably resides here:
> [image]
> `viewUpdater.js` line 998 to 1015
> 
> tool element is not found

Indeed, when the status is 'Offline' there is no tool element

NotExpectedYet, 2021-04-19:
That's a pointless if statement if I ever saw one 😁

NotExpectedYet, 2021-04-23:
Still couldn't reproduce this but looking now.

Spool selected. 

Index 0 printer offline
index 1 printer printing. 

XD

NotExpectedYet, 2021-04-23:
PR #447 

Once david has time to review that PR I'd like confirmation from you @3DRemake if at all possible. 

I'm not sure if the issue lies deeper with something else as if you check the PR I tried a few different variations from my knowledge of the backend/frontend. I've put checks in and removed some of the pointless code and considering the printer is offline this should suffice to resolve the issue for now.

3DRemake, 2021-04-24:
Tested on development branch. Issue is fixed

closed 2021-04-24 by davidzwa

NotExpectedYet, 2021-04-24:
> Tested on development branch. Issue is fixed 

Thank you for the report back! Good to know.