OctoFarm/OctoFarm #448: Can't save database URI from new UI.

NotExpectedYet, 2021-04-23:
### To better help the developer understand your issue, please fill in the information below. Failing to do so will waste the developer and your own time in sorting the issue. Thanks!

**Client or Server side error** 
Server

**Describe the bug**
Didn't notice this before because my old config/db.js folder was still in play, and now for some reason that doesn't make a difference. I can't create the .env file from the UI, has to be done manually. It's also deleting out my config/db.js folder that I add back in, and still telling me I have database issues. 

Funnily enough I'm using this as my url: MongoURI: "mongodb://127.0.0.1:27017/octofarm"

**To Reproduce**
Steps to reproduce the behavior:
1. Create config/db.js folder and file
2. start the service `npm start`
3. Load the interface to be told of a database warning. 
4. Test works fine, but unable to save when I don't already have a .env file. 

**Expected behavior**
To be able to create the database file from the UI 

**Screenshots**
[image]

**Versions (please complete the following information):**
 - NodeJS Version [e.g. 13]: 14.6.1
 - OctoFarm Version [e.g. 1.5.5.7] 1.1.13-development

NotExpectedYet, 2021-04-23:
MONGO environment variable is not set. Assuming default: MONGO=mongodb://127.0.0.1:27017/octofarm

Weirdly that's just started working. Nothing wrong with my mongodb service either. hmmm

NotExpectedYet, 2021-04-23:
```
throw new Error("Could not parse current .env file. Please ensure the file contains lines with each looking like 'OCTOFARM_PORT=4000' and so on.");
    ^

Error: Could not parse current .env file. Please ensure the file contains lines with each looking like 'OCTOFARM_PORT=4000' and so on.
    at Object.writeVariableToEnvFile (/home/s1mpleman/Development/NotExpectedYet Development/octofarm-development/OctoFarm/server_src/utils/env.utils.js:48:11)
    at ensureMongoDBConnectionStringSet (/home/s1mpleman/Development/NotExpectedYet Development/octofarm-development/OctoFarm/app-env.js:96:16)
    at setupEnvConfig (/home/s1mpleman/Development/NotExpectedYet Development/octofarm-development/OctoFarm/app-env.js:129:3)
    at Object.<anonymous> (/home/s1mpleman/Development/NotExpectedYet Development/octofarm-development/OctoFarm/app.js:17:1)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47
[nodemon] app crashed - waiting for file changes before starting...

```

That was a test I just made with a custom database URL in config/db.js

NotExpectedYet, 2021-04-23:
Works fine if I've already got a .env but as I'm sure your aware that ain't gonna be anyone when this release hits.

davidzwa, 2021-04-24:
5 pushups for me

closed 2021-04-24 by NotExpectedYet