OctoFarm/OctoFarm #466: Rate limit the reminder for update more

NotExpectedYet, 2021-04-25:
- Description
Annoying! XD

- Repro

davidzwa, 2021-04-25:
As the updater was in limbo state it could not update as we didnt have OCTOFARM_ALLOW_PRERELEASE_INSTALL defined as env variable, but constantly thought 'hey but prerelease is out'!

So the rate limiting... mark read should work fine now. Can always spend time on doing it, but I wonder if its really needed.
If we look at OctoPrint they spam you until you hit mark read anyway.

NotExpectedYet, 2021-04-25:
> As the updater was in limbo state it could not update as we didnt have OCTOFARM_ALLOW_PRERELEASE_INSTALL defined as env variable, but constantly thought 'hey but prerelease is out'!
>
> So the rate [...]

Spamming till mark read is fine, but yeah tested and re-produced the fixes fine all golden.

closed 2021-04-27 by NotExpectedYet