OctoFarm/OctoFarm #479: Struggling through Ubuntu Server on a RPI4

Polloman93, 2021-04-26:
Hi all.
I'm new here. It's the first time I'm using GitHub at 100%. 

Octofarm save my live for 4 months in Windows but it started to go laggy and I decided to get into ubuntu server and a Raspberry. I started with a Model 3B+ but, with 30 printers, didn't work properly.

I'm now with a RPi4. 2 days ago I have octofarm running on it, but it wasn't as stable as i wish (my linux knowledge is stilll pretty short). So, today I started over and everything was pretty straight forward until this:

[image]

Fresh install following the new guide (amazing btw)

Octofarm version: 1.1.13
Node Version: 12.22.1
Mongodb version 4.4.5
Operating system: Ubuntu Server on a RPi4

[image]

This is a pm2.error.log

I really dont know how to do because, as I said, Ik pretty beginner in code, ubuntu...

Thanks for the help in advance!

davidzwa, 2021-04-26:
Please make sure to update to Node 14 ;) This is a requirement.
https://nodejs.org/en/
Here you could read why:
https://github.com/OctoFarm/OctoFarm/issues (this is pinned)
For installation instructions:
https://octofarm.net/installation (not 100% clear, but its complete)

Note: 1.1.13 is still in **prerelease**, which means we intend to make this error clearer for new users, so keep that in mind! We plan to release 1.1.13 this week.

Happy printing!

closed 2021-04-26 by davidzwa

NotExpectedYet, 2021-04-26:
Thanks for pointing that out @Polloman93, I missed that! 

Also someone who reads the instructions XD. 

Updated the site thanks again.

davidzwa, 2021-04-26:
Much obliged, we updated the site

Polloman93, 2021-04-26:
As I said, I'm beginner in this so the instructions are a must for me.  The optimal thing is to read it completly not just a line and take that as the TRUTH xDD

Nice work guys, this is a really improvement to poeple with more than 3 printers. 

Thanks for all.

davidzwa, 2021-04-26:
As a dev its sometimes hard to see the obvious things, any way we can improve the docs from your point of view?

Polloman93, 2021-04-26:
Docs are perfect. Now are less confusing and for people like me the copy button add is the key.