OctoFarm/OctoFarm #483: Release and pre-release work flow with PRs commited.

NotExpectedYet, 2021-04-26:
- Description
Update the workflows to automatically include the changelog inside of the release. This can be built from the PR's and include the PR user from such. David will have to stop rebasing my PRs  though as he will steal them all 🤣

Should follow the current changelog format and we could probably use folders like we do we bug fix/hotfix/feature to group them or tags, not sure how that works with PRs. 

- Repro

davidzwa, 2021-06-26:
Maybe 'borrow' how to tackle this from big repos? Maybe OP has a good way?

Orrrrr we continously update the Changelog.md with users and write the intro story as we go

NotExpectedYet, 2021-06-26:
I've seen a mix of people using GitHub actions / the keep a changelog actions based off of PR notes.

NotExpectedYet, 2021-06-26:
https://github.com/marketplace/actions/release-changelog-builder

That gets us an automated change log which would be nice.