OctoFarm/OctoFarm #513: Not showing all printers

Mat1j, 2021-04-29:
### Versions

* Your **OctoFarm version** (latest 1.1.13): 1.1.13
* Your **OctoPrint version** (latest 1.5.3): 1.5.3
* Docker or pm2: pm2
* **NodeJS version** 14 or 15 (not for docker): 14
* Operating System (f.e. Raspberry Pi OS, Ubuntu, Windows): Ubuntu on RaspberryPi
* OctoPrint Plugins (e.g. bed visualiser, octoklipper, printtimegenius): none

### Reproduction

* **Location** of the problem (choose Server, NodeJS, MongoDb database or local website)
* Description of the problem: I am not able to control nor see some of my printers. I'm not even able to connect them again, course their IPs are already in database. Their IPs are pinged successfully, but printers not showing in printers menu.
* Steps needed to **reproduce** the problem. not sure :/
* Expected behavior:
* (Optional) To avoid the problem I had to:

### OctoFarm Logs

If you've got access to the System screen it's better to just upload the log dump file as it will
contain all relevant information for the developers.
[octofarm_dump (1).zip](https://github.com/OctoFarm/OctoFarm/files/6399060/octofarm_dump.1.zip)
[image]
[image]
[image]

1) **OctoFarm Server logs** Add the system page logs .zip file (if you cant access it, upload your `./logs` folder content here)
2) **Console logs** in your browser.
   (How to: https://javascript.info/devtools)
3) **Screenshots**
   These are always effective, add screenshots to help explain your problem.

NotExpectedYet, 2021-04-29:
Epic thank you for the quick report. I've got some time in the morning so I will have a look for you.

NotExpectedYet, 2021-04-29:
If you could screenshot the console page from your browser as well that would help narrow it down :)

Mat1j, 2021-04-29:
> 
> 
> If you could screenshot the console page from your browser as well that would help narrow it down :)

sure, i hope got the correct console 
[image]

NotExpectedYet, 2021-04-29:
> > If you could screenshot the console page from your browser as well that would help narrow it down :)
> 
> sure, i hope got the correct console
> [image]

Sorry no, it's the browsers console window here: 
[image]

You press "F12" whilst on the Printers page and then click the little console tab. 

Will just help me narrow it down to client or server side more. :)

Mat1j, 2021-04-29:
> Sorry no, it's the browsers console window here:
> [image]
>
> You press "F12" whilst on the Printers page and then click the little console tab.
>
> Will just help me narrow it down to client or server [...]

[image]

NotExpectedYet, 2021-04-29:
Just checked your logs, that's weird the file key is coming back as null. Will be causing the printerState error you just posted. 

Should be able to sort it out for you to at least view those files again. That's enough to go from thank you very much!

NotExpectedYet, 2021-04-30:
Could I have PR'd an attempted fix for this. Just waiting for david to review it. 

I'm not entirely sure what's gone on here, there we're some errors stating you couldn't sync the file list. Which is usually caused by large file lists &/Or WiFi'd OctoPrints. You also seem to be running multiple instances from the same Pi's which could cause API/Scanning issues over WiFi. I tried reproducing here but I couldn't, even with a file list missing and getting no information about the job from the current print so a bit baffled to what's gone on at the moment. 

It's up to you if you want to see if the issue has been resolved by the PR but I believe deleting those printers and re-adding them should fix your issue. That's if you can see them on the Delete option anyway. It doesn't reference the printerState. 

Whatever it is is happening after a re-scan as well.

Mat1j, 2021-04-30:
> I have PR'd an attempted fix for this. Just waiting for david to review it.
>
> I'm not entirely sure what's gone on here, there we're some errors stating you couldn't sync the file list. Which is [...]

Well i finally solved it, or at least i make it work.
I nuked the printers database (was not able to delete printers before), 
and now i can add remove and i see all my printers again. 

Thanks  very much for all your help and effort, apart from this OF saves me ton of time :) .

closed 2021-04-30 by Mat1j

NotExpectedYet, 2021-04-30:
> Well i finally solved it, or at least i make it work.
> I nuked the printers database (was not able to delete printers before),
> and now i can add remove and i see all my printers again.
>
> Thanks very [...]

No worries! 

I did see some issues with scanning the full file list on your instances but apart from that everything else looked fine so was a confusing one. Glad your up and running anyway. Thanks for letting us know.