OctoFarm/OctoFarm #515: Not able to add printer

Nomad965, 2021-04-29:
### Versions

* Your **OctoFarm version** (latest 1.1.13): 1.1.13
* Your **OctoPrint version** (latest 1.5.3): 1.6.0
* Docker or pm2: pm2
* **NodeJS version** 14 or 15 (not for docker): 14.16.1
* Operating System (f.e. Raspberry Pi OS, Ubuntu, Windows): Windows 10 Pro 64bit / OSBuild 19042.928
* OctoPrint Plugins (e.g. bed visualiser, octoklipper, printtimegenius): OctoDash, PrusaThumbnail

### Reproduction

* **Location** of the problem (choose Server, NodeJS, MongoDb database or local website) webgui
* Description of the problem: Cannot add new printer
* Steps needed to **reproduce** the problem. Register new user> setup new printer> No printer is available
* Expected behavior: Should be able to add new printer, browser brave and firefox same issue
[image]
[octofarm_dump.zip](https://github.com/OctoFarm/OctoFarm/files/6402123/octofarm_dump.zip)

* (Optional) To avoid the problem I had to:

### OctoFarm Logs

If you've got access to the System screen it's better to just upload the log dump file as it will
contain all relevant information for the developers.

1) **OctoFarm Server logs** Add the system page logs .zip file (if you cant access it, upload your `./logs` folder content here)
2) **Console logs** in your browser.
   (How to: https://javascript.info/devtools)
3) **Screenshots**
   These are always effective, add screenshots to help explain your problem.

NotExpectedYet, 2021-04-30:
Only the first user created can create printers. New users get the user key and cannot. There's a note on the register screen.

[image]

Nomad965, 2021-04-30:
Hi James, thank you for the prompt reply, I only have one user set and not other user created.

NotExpectedYet, 2021-04-30:
Hello! Your welcome. 

Hmm I tried to reproduce your issue before posting and my Administrator user is correctly created. 

There's a .env file inside your OctoFarm folder. Could you please try just appending some extra text onto the MONGO url something like the example below. This will create a new database. 

If the problem persists I'll probably need some of the logs out of your OctoFarm/logs folder.

```
MONGO=mongodb://127.0.0.1:27017/octofarmnew
```

davidzwa, 2021-04-30:
Screenshots in discord, logs of current situation were attached by O P

NotExpectedYet, 2021-04-30:
There's nothing in the logs sorry didn't notice you'd posted them had just woken up aha. Hopefully a new database resolves the issue.

davidzwa, 2021-04-30:
James showed me the code where the error occurs, and I was able to get the same error as you @Nomad965. The problem is clear: the Administration group is not set to your user.

We have some possible ideas:
- does a new database fix it?
- could it be that 2 users were created per accident? Can you download https://www.mongodb.com/try/download/compass and check your octofarm users table?

Nomad965, 2021-04-30:
Could this be the problem ?
Just delete one ?

[image]

Nomad965, 2021-04-30:
Could this be the problem?

Just delete all of the users and redo again or just delete the non admin user ?
[image]

NotExpectedYet, 2021-04-30:
> Could this be the problem?
> 
> Just delete all of the users and redo again or just delete the non admin user ?
> [image]

Certainly is! Delete the user and you should be fine :) 

Thanks for getting back to us.

Nomad965, 2021-04-30:
Ahhhhaa, now I see what went wrong, a mistake on my behalf, looks like I registered the Admin account under the username Nomad, when I tried to login the first time round I, used nomand with lowercase "n". 

The db does not identify the difference between uppercase and lowercase admins and treats them as one. So thinking when I did the 1st time registration, I did not pay attention to that, and registered again a new user with nomad, not knowing that the right username for admin was Nomad.

Just logged in now, and proceeding with the rest of the setup and testing.

Thank you very much for the help and support @NotExpectedYet @davidzwa 

Cheers..

davidzwa, 2021-04-30:
This is awesome news Nomad. We know what to improve and you know how to work around it.

Win win!

Nomad965, 2021-04-30:
Just something else I noticed not that it caused any login problems or authentication, when creating the api key under application / octoprint interface the assigned user is nomand with lower case n and the Admin user in octofarm is with capital N for nomad, the octofarm accepted the API key without any issues.

NotExpectedYet, 2021-05-01:
> Just something else I noticed, not that it caused any login problems or authentication, when creating the api key under application / octoprint interface the assigned user is nomad with lower case n and the Admin user in octofarm is with capital N for nomad, the octofarm accepted the API key without any issues.

Those users are un-related. One is for OctoPrint and one for OctoFarm, they can be completely different.

Nomad965, 2021-05-01:
Well noted, thank you for the clarifications.

NotExpectedYet, 2021-07-01:
Bad stale bot

closed 2021-07-11 by NotExpectedYet