OctoFarm/OctoFarm #526: Printer map

davidzwa, 2021-05-01:
## Description

We have to think about:
- cleanup the printer map (davidzwa fork)
- define a predefined grouping system (blocks of 3x2, 2x2, 1x2, 1x1 etc, or none at all)
- (optional/alternative) drag and drop system to organize the blocks
- define a gallery system (multiple blocks)
- tabs with index numbers to switch between 'floors'
- printer group collection being used instead of single-value printer labels
- adjusting the tables to use these groups
- make the printer naming and URL settings 100% robust

davidzwa, 2021-05-03:
@NotExpectedYet I dont think this is going to be ready soon as it is dependent on webpack (not a hard requirement, but it is very beneficial) and some backend cleanup work.

I would propose to move it to 1.2, what do you think?

NotExpectedYet, 2021-05-03:
> @NotExpectedYet I dont think this is going to be ready soon as it is dependent on webpack (not a hard requirement, but it is very beneficial) and some backend cleanup work.
> 
> I would propose to move it to 1.2, what do you think?

Yeah no worries at all man, shift it into that project all good.

NotExpectedYet, 2021-07-03:
Stale away!

NotExpectedYet, 2021-07-11:
eurgh

closed 2021-09-26 by github-actions[bot]