OctoFarm/OctoFarm #532: OP printer settings dialog does not query fresh settings

davidzwa, 2021-05-02:
- Description
Upon editing settings we gotta have the latest settings set from OP. Fix that.

- Repro
Open printer settings dialog.

davidzwa, 2021-05-09:
It now queries the settings, but its not passed on to the UI.

```
09/05/2021, 10:09:18 | INFO | OctoFarm-State | Attempting to connect to API: api/system/commands | http://192.168.1.177 | timeout: 10000
09/05/2021, 10:09:18 | INFO | OctoFarm-State | Successfully grabbed System Information for...: http://192.168.1.177
09/05/2021, 10:09:35 | INFO | OctoFarm-State | Attempting to connect to API: api/settings | http://192.168.1.177 | timeout: 10000
09/05/2021, 10:09:35 | INFO | OctoFarm-State | Detected Pi Support!
09/05/2021, 10:09:35 | INFO | OctoFarm-State | Attempting to connect to API: api/plugin/pi_support | http://192.168.1.177 | timeout: 10000
09/05/2021, 10:09:35 | INFO | OctoFarm-State | Got from endpoint: data:{"model":"Raspberry Pi 4 Model B Rev 1.2","model_unrecommended":false,"throttle_state":{"current_issue":false,"current_overheat":false,"current_undervoltage":false,"past_issue":false,"past_overheat":false,"past_undervoltage":false,"raw_value
":-1}} |
```

NotExpectedYet, 2021-05-11:
> It now queries the settings, but its not passed on to the UI.
>
> ```
> 09/05/2021, 10:09:18 | INFO | OctoFarm-State | Attempting to connect to API: api/system/commands | http://192.168.1.177 | timeout: [...]

Should be able to sort without killing anything major. It just needs to call the cleaner again on the system one, or update the cached. Think I have a graceful work around after your comments anyway.

NotExpectedYet, 2021-05-13:
Wasn't a cache update issue, it ran perfectly fine. It was me not pulling the correct information. Port preference comes from my getState command, so it needed to fire: getState, getSettings, getSystem. 

Has been fixed in latest PR. Just need to sort the button and write the test now.

davidzwa, 2021-05-13:
Nice job! What in PrinterClean does it call, like a whole refresh?

NotExpectedYet, 2021-05-13:
I shouldn't have changed printerClean here? If I did I'll undo it as I shouldn't need to. 

PrinterClean.generate function, it will run the cleaner for a single printer you pass to it so it now reruns that. In my tests the 500ms timer had mostly run already like by the time the updated settings had been called it was just to make sure.

NotExpectedYet, 2021-05-13:
No your right I did aha. I tried to use the cache file, is that complete? It kept returning undefined for me so I updated the printer function to be able to return a single printer as well as the whole bunch.

davidzwa, 2021-05-13:
Ah my goof there, no that's a work-in-progress :S

commit a76df7da (2021-05-03): Initial Printer Settings class refactor. Fixed #532
commit 6467c2bc (2021-05-13): Fixed #532 - Now correctly updates octoprint settings
commit 81dcb7bf (2021-05-29): Feature/printer settings dialog improvements (#545)

* version 1.1.13-hotfix

* remove useless code

* Initial Printer Settings class refactor. Fixed #532

* seperated try/catches

* missing script import

* #Fixed 351

* add power reset listener

* button... doing... nut in

* Squash printer settings dialog

* Fixed #532 - Now correctly updates octoprint settings

* Fixed #576
Fixed #577

* Fix for event listener not getting applied to save button

* added checks for octofarm successful/unsuccessful update

* allow Click Me button to refresh settings page.

* clean up and test init

* Add aut [... cut]
closed 2021-07-11 by NotExpectedYet