OctoFarm/OctoFarm #538: Filament / History Graphs stay "loading..." when no data presented if outside 30 day window.

NotExpectedYet, 2021-05-02:
### Description of the problem

```bash
Not real dire, as depends how active someones farm is. Data is outside of 30 day window graphs still show "loading..."
```

### Reproduction of the problem

```bash
Go to graphs without that data in a 30 day window.
```

NotExpectedYet, 2021-07-02:
@davidzwa this was fixed no?

davidzwa, 2021-07-02:
I dont think it is. I was mainly busy with the data inside the 30 day interval. You were able to spot this bug outside it and I dont recall us fixing it

NotExpectedYet, 2021-07-02:
> I dont think it is. I was mainly busy with the data inside the 30 day interval. You were able to spot this bug outside it and I dont recall us fixing it

I'll double check.

NotExpectedYet, 2021-07-03:
[image]

Yeah fixed, well outside the 30 day window

davidzwa, 2021-07-03:
Uhh or 90? Not sure right now

NotExpectedYet, 2021-07-03:
> Uhh or 90? Not sure right now

Is it still 90? 

that's from dashboard, was taking it on face value you updated the variable name in the re-factor. If that's not the case then could still be a bug xD

[image]

NotExpectedYet, 2021-07-03:
Think the dashboard charts need a check over. Get this when enabling the bottom ones. 

[image]

davidzwa, 2021-07-03:
Yeah it was hard for me to test them. Rushed it a bit

NotExpectedYet, 2021-07-03:
We have errors now though! 😁

NotExpectedYet, 2021-07-04:
[image]

Yep still brokey, need's to use the actual date ranges that are created, not -90 days haha.

closed 2021-09-27 by github-actions[bot]