OctoFarm/OctoFarm #541: Allow Admin users to view, edit and create users.

NotExpectedYet, 2021-05-03:
### Description of the problem

```bash
Currently you can't create users without Registration been turned on. This is a limitation for any admin wanting to control their backend. 

1. Allow Admins view a list of users
2. Allow Admins to edit the list of current users.
3. Allow admins to create new users on the spot and assign them a password. 

Possible but not required:
1. Allow the setup of SMTP credentials for password reminders
```

### Reproduction of the problem

```bash
...
```

davidzwa, 2021-05-03:
I need to refer you to this:
https://github.com/OctoFarm/OctoFarm/discussions/528

Has some value maybe.

NotExpectedYet, 2021-05-12:
@davidzwa certainly does was just adding it as an actual ticket as it's something I've been meaning to add for a while.

NotExpectedYet, 2021-07-12:
No stale crazy bot

closed 2021-10-03 by github-actions[bot]