OctoFarm/OctoFarm #555: `Offline after error` is still marked as connected in OctoFarm

davidzwa, 2021-05-07:
### OctoFarm Version

1.1.14-rc1 with webpack

### OctoPrint Version(s)

1.6.0

### pm2 or docker?

docker

### Operating System

Win10

### OctoPrint Plugins

None

### Location of the problem

NodeJS

### Description of the problem

```bash
When connecting to a printer which is having `udev` conflict issues OctoPrint says 'offline after error'. That's a problem of its own, but I would expect OctoFarm not to show green then.

Both the printers page and the printer map show that the printer is connected instead. That's a bug probably.
```

### Steps taken to reproduce

```bash
1. Setup finicky `udev` rule (still figuring this out...)
2. Rescan ports on OctoPrint
3. Port detected
4. Port connect or AUTO connect fails with 
> 'State: Offline after error No more candidates to test, and no working port/baudrate combination detected.'
5. Try the same on OctoFarm, will say it is connected and green.
...
```

### What did you expect to happen

```bash
Not connected label on OctoFarm printer management.
```

### To avoid the problem I had to

```bash
Make an issue and be amazed by this beautiful form without file uploads
```

davidzwa, 2021-05-07:
The reason I mention this is because 'printing from SD' is also yellow, so Ive come to associate yellow with being OK.
[image]

Finally, white on yellow isnt readable for this matte MTB3D trash 1-pixel screen

NotExpectedYet, 2021-05-08:
I never supported printing from SD as I've never used it. I gather it's just falling back to a default. Yellow is generally warning.

davidzwa, 2021-05-08:
Hmm mixing printing from SD and USB connection is a small hell. I've had to throw away prints because of (I think) my mistakes - I still have to figure out where this is caused.
Its a migration thing, a company must keep printing while setting up OctoFarm. I guess I should find more suitable moments where all the printers are done or offline.

Practical todo's
- The label for Offline after error could be changed to red
- The 'connection' switch is green, should be set to disconnected

davidzwa, 2021-05-30:
### Description of the problem

```bash
Just some proactive coding we have to do as Websocket is in a temporary state (guess, pls validate before acting on it)
```

### Reproduction of the problem

```bash
Printer online
Connect OP
Power off printer
See yellow "Offline after error"
Refresh printer
Red "Disconnected" shows

The last 2 points can be acted on proactively immediately.
```

NotExpectedYet, 2021-05-30:
So this was basically because I hadn't know about that error state. gathered it's new in 1.6.0. Before all errors started with a "Error!" string so I gather it's Gina trying to be more user focused in the error responses. Anymore crop up it's pretty easy to add the detection as you'll see on the PR. It's not really necessary in the websocket section apart from error's. From OP state as I override/change a few of those. IE, printer says "Disconnected" when you first boot up octoprint, then "Offline" after you connect and disconnect which seemed confusing to me so "Offline" -> "Disconnected" as well ha.

commit 381dc0c3 (2021-05-31): Bugfix/555 offline after error (#614)

* version 1.1.13-hotfix

* Fixed #555 - Offline after error

* fixed the overwrite lag
commit 5ca8da40 (2022-05-19): chore(main): prep for v1.2 release - strong and stable (#946)

* Bugfix/609 printer bulk functions broken (#610)

* version 1.1.13-hotfix

* wow what a dodo

* Fixed #609

* split function for printer info

* Add tests, remove await and annotate TODO

* Forgot to clean up

* Bugfix/555 offline after error (#614)

* version 1.1.13-hotfix

* Fixed #555 - Offline after error

* fixed the overwrite lag

* V2/add job stats (#619)

* Convert V1 job statistics

* Apply tests from V1

* Job statistics fixed

* 574 state split (#575)

* Squash state op api refactor

* Apply compat check to new OP clien [... cut]
closed 2021-07-11 by NotExpectedYet