OctoFarm/OctoFarm #566: Change git pull/fetch mechanism over to tags

NotExpectedYet, 2021-05-12:
### Description of the problem

```bash
Currently the update mechanism uses git fetch and pull to update the local branch. This can be impoved with checkout / fetch of specific git tags from the updater function. 

Current proposition with simple git.

`git.fetch(origin, ${branch}, ['--rebase': 'true', 'tag': '${tag_number}']` (fetch remote branch to local)
Note: I gather the rebase option is better for production installs? Or should we stick with merge here? 

`git.checkout({tags/${tag_number})`

Would need to confirm and test these, just noting it down for now.
```

davidzwa, 2021-05-12:
And what about 'git fetch' and then 'git reset --hard origin tag_name' or something syntactically correct?
Users should not have file changes, nor commits.
Oh and dev mode should be rock solid if we do this

NotExpectedYet, 2021-05-12:
@davidzwa it checks for those anyway, I gather these are two ways to skin the same cat no? I was been specific with the fetch to reduce what was pulled but I don't know enough about git which would be better.

davidzwa, 2021-05-12:
What the git reset will do is completely cancel out any changes, so it the hardest override out there. It will absolutely run over any uncommitted change.

closed 2021-09-26 by github-actions[bot]