OctoFarm/OctoFarm #567: PrinterClean `x of undefined` when you start a new database, don't add printers and restart

NotExpectedYet, 2021-05-12:
### Description of the problem

```bash
Currently when you start a new database on development then you get a unhandled error event as it's trying to find 
`heatMap[i].data[lastInArray].x` when `(heatMap[i].data[lastInArray]` doesn't exist.
```

### Reproduction of the problem

```bash
...
```

davidzwa, 2021-05-12:
We've come to the conclusion that the problem lies in a race-condition caused by the code at the bottom of `index.js` in combination with the code in `app-core.js:serveOctoFarmNormally`.

The bottom of index.js contains code which runs in a 2500 ms loop, but apparently isnt done quickly enough on all systems or it is skipped because mongoose is not ready to connect yet.
The code calls `PrinterClean.sortCurrentOperations(printersInformation);` which calls `PrinterClean.Heatmapping` after a certain counter only, but this counter is 17280 by default (run at start). So all in all combined can result in heatmap being incorrectly saved to database before using it in `await PrinterClean.initFarmInformation();`

NotExpectedYet, 2021-05-12:
Good investigation. 

I gather we should open another ticket for reorganising the timers and what not for start up and beyond.

closed 2021-07-11 by NotExpectedYet