OctoFarm/OctoFarm #577: sumValuesByGroup undefined

NotExpectedYet, 2021-05-15:
### Description of the problem

```bash
UnhandledPromiseRejectionWarning: ReferenceError: sumValuesGroupByDate is not defined

Have fixed in #545
```

### Reproduction of the problem

```bash
Open statistics on latest dev
```

davidzwa, 2021-05-15:
Cannot reproduce this. Which file does this error occur in?

NotExpectedYet, 2021-05-15:
> Cannot reproduce this. Which file does this error occur in?

printerClean, might be a change I made on the printer settings branch. Dunno, just noticed it. Fixed it anyway, just needed "this." adding.

davidzwa, 2021-05-15:
PrinterClean.sumValuesByGroup seems better to me, as it indicates that its static explicitly.

NotExpectedYet, 2021-05-15:
> PrinterClean.sumValuesByGroup seems better to me, as it indicates that its static explicitly.

Is there a difference?

davidzwa, 2021-05-15:
`this` can become wrongly defined for certain callback based code in static contexts and so sometimes care is needed. `PrinterClean.` can never go wrong.
If we ever change this class to be non-static, only `this.sumValuesByGroup` will work and not `PrinterClean.sumValuesByGroup`, so this is also a matter of being 100% clear in our code.

Dont get me wrong, in V2 it's all `this.` based, but this code isnt.

Btw, I will PR this separately to add tests, changelog and the fixes.

PR #578 (2021-05-15): Two bugs solved apart from printer settings
- [x] Fixes #577 and #577 explicitly
- [x] Changelog
commit 5a0b93a2 (2021-05-15): Fixed #576
Fixed #577
commit 81dcb7bf (2021-05-29): Feature/printer settings dialog improvements (#545)

* version 1.1.13-hotfix

* remove useless code

* Initial Printer Settings class refactor. Fixed #532

* seperated try/catches

* missing script import

* #Fixed 351

* add power reset listener

* button... doing... nut in

* Squash printer settings dialog

* Fixed #532 - Now correctly updates octoprint settings

* Fixed #576
Fixed #577

* Fix for event listener not getting applied to save button

* added checks for octofarm successful/unsuccessful update

* allow Click Me button to refresh settings page.

* clean up and test init

* Add aut [... cut]
closed 2021-07-11 by NotExpectedYet