OctoFarm/OctoFarm #586: category undefined error printer management

davidzwa, 2021-05-18:
Undefined line 50

NotExpectedYet, 2021-05-18:
> Undefined line 50
> 
> Repro:
> click on printer control button for a printer which was newly created. And connected just fine.
> Dialog spinner keeps loading with the error in the title.
> 
> Probably some transient error, page reload does not fix it.
> 
> ```
> printerManager.js:50 Uncaught (in promise) TypeError: Cannot read property 'category' of undefined
>     at printerManager.js:50
>     at Array.forEach (<anonymous>)
>     at Function.init (printerManager.js:49)
>     at HTMLButtonElement.<anonymous> (printerManagementRunner.js:3293)
> ```
> 
> The code
> 
> ```
> printerControlList.forEach((list) => {
>           if (list.state.category !== "Offline") {
>             printerDrop.insertAdjacentHTML(
>               "beforeend",
>               `
>                   <option value="${list.printerID}" selected>${list.printerName}</option>
>               `,
>             );
>           }
>         });
> ```

That printer list comes from PrinterClean so I gather it's something with the recent changes?

davidzwa, 2021-05-18:
So this docker container (davidzwa/octofarm) was built 11 days ago, before PrinterClean and HistoryClean were touched. It was based on 1.1.13-hotfix + webpack back then.
PrinterClean was not touched a lot. It was HistoryClean which was refactored.

Will rebuild the container and debug more either way.

NotExpectedYet, 2021-05-18:
> So this docker container (davidzwa/octofarm) was built 11 days ago, before PrinterClean and HistoryClean were touched. It was based on 1.1.13-hotfix + webpack back then.
> PrinterClean was not touched [...]

That seems like a issue that should crop up a lot more and haven't seen it personally was all so no idea at minute XD. Test away ^^ I'll try repro over here when I get chance at weekend.

davidzwa, 2021-05-18:
Cant repro now that the server has started, so either its gone or its a transient error.

Condition:
- The printer URL was incorrect (IP address input wrong by me)

Some leads or related weirdness:
- The printer name I added was overwritten and changed back to IP **when creating the printer** 
==> why does this happen (still) when name is not empty and should not be changed back to an IP address?
- The printer URL didnt update until I refreshed the printer (after I saved new url in printer settings dialog)
==> Probably related to settings refresh.

NotExpectedYet, 2021-05-18:
> Cant repro now that the server has restarted and its now on dev+webpack, so either its gone or its a transient error after creating adding printers with bad/incorrect URLs. So we should try and reproduce it that way (a bad printer URL which times out).
> 
> Condition:
> 
> * The printer URL was incorrect (IP address input wrong by me)
> 
> Some leads or related weirdness:
> 
> * The printer name I added was overwritten and changed back to IP **when creating the printer**
>   ==> why does this happen (still) when name is not empty and should not be changed back to an IP address?
> * The printer URL didnt update until I refreshed the printer (after I saved new url in printer settings dialog)
>   ==> Probably related to settings refresh.

Hmmm, wonder if it's server side then with an incorrect printer not firing something on the cleaner. 

The first point sounds like an incorrect check for printer name, it shouldn't be overwriting it when a name is entered it should be using it. As far as I remember I just check for a blank string there. 

If you update the printer URL it should do the refresh there and then. Which box did you save it in? The bulk or the actual printer settings? If it's printer settings I don't think that actually refreshes the printer the same, the bulk one certainly does. I could probably add that into my current PR tbh. Probably wasn't initially considered when writing the printer settings box. If it's needed a re-scan then would certainly be that thats the case. Not a hard fix, `if ip/port is changed rescan printer`

NotExpectedYet, 2021-05-18:
The first one is in the "GetSettings" check

Second one, can sort in that PR no drama. Will open a ticket for it.

NotExpectedYet, 2021-05-18:
#588
#587

NotExpectedYet, 2021-05-18:
Infact I'll do another PR for it. xD

davidzwa, 2021-07-31:
This is irrelevant now. Marking as fixed.

closed 2021-10-16 by davidzwa