OctoFarm/OctoFarm #617: Resyncing printer shows "Shutdown" shortly

davidzwa, 2021-05-31:
### OctoFarm Version

1.1.14-rc1

### OctoPrint Version(s)

1.5+

### pm2 or docker?

pm2

### Operating System

Pi OS 

### OctoPrint Plugins

Default

### Location of the problem

Not sure

### Description of the problem

```bash
Resyncing printer shows "Shutdown" API state shortly before returning to "Connected" API state.
```

### Steps taken to reproduce

```bash
1. Resync
```

### What did you expect to happen

```bash
Connected > Loading > Connected
```

### To avoid the problem I had to

```bash
File a bug report in this awesome form
```

NotExpectedYet, 2021-06-19:
This is because the Host state and the OP state are coupled at the moment. 

The host state should be separated in state.js really. The host state get's decided from the API calls, where a ping or another method to the host IP would be much better. If OP returns an error on the API when it's booting for example, that will cause the host state to come back "Shutdown".

davidzwa, 2021-08-19:
Irrelevant now with canary

closed 2021-08-19 by davidzwa