OctoFarm/OctoFarm #618: History Page is not showing any information.

asylumlife, 2021-06-01:
### OctoFarm Version

1.1.13

### OctoPrint Version(s)

1.6.1

### pm2 or docker?

pm2

### Operating System

Ubuntu on Raspberry Pi

### OctoPrint Plugins

themify, Tab order, SpoolManager, Software update, slicer thumbnails, prusa leveling guide, prusa ETA override, Pi Support Plugin, Octopod, octolapse nabber temperature, logging, ironconnect, gpio control, code viewer, floating nabber, firmware updater, Firmware check, file check, filament manager, error tracking, draggable files, discovery, cost estimation, core wizard, cancel object, backup and restore, application keys plugin, anonymous usage tracking, announcement plugin, action command prompt support, Action command notification support

### Location of the problem

Web Interface

### Description of the problem

```bash
...
I have been running the Octofarm for over 54 days. Prior to the latest upload I had a history page that should the data it should. now it just keeps telling me to print more.
```

### Steps taken to reproduce

```bash
1. Glick on Histor
2. Look :-)
```

### What did you expect to happen

```bash
...
To see the history data
```

### To avoid the problem I had to

```bash
Not sure what to do...
```

asylumlife, 2021-06-01:
[octofarm_dump (1).zip](https://github.com/OctoFarm/OctoFarm/files/6579890/octofarm_dump.1.zip)
[image]
[image]

davidzwa, 2021-06-01:
We will get to the bottom of it. The easy fix is already in place, next we need to find and resolve the cause.

NotExpectedYet, 2021-06-02:
@davidzwa so the information cleaner would suggest it's the file property been null. History logs don't show that failing at all though. 

I've noticed before a weird edge case where OF doesn't get the job key where file resides so could be that. Wonder if it's from a quicky started then cancelled print maybe coupled with that onFailure error. Just a guess at the moment like. 

@asylumlife if you go to the system page and then "Server" section and hit save it will clear up that error for capturing failed prints. We've got some fixes for that issue coming. If it wasn't capturing failed prints anyway that's why. The file not been there is weird as you wouldn't have seen it in the UI if that was the case. We are on it though just to reiterate David's words.

asylumlife, 2021-07-05:
I have tried going to the server settings area several times and hitting save but it has never resolved the issue.

NotExpectedYet, 2021-07-05:
> I have tried going to the server settings area several times and hitting save but it has never resolved the issue.

Sorry it's not this issue you linked that would be resolved by that. It's just an error in your logs that might have stopped failed prints capturing in the first place. Your job issue is different. As david said we have a fix on the now titled V1.2-Rc1 so if you get chance it would be good to see if it allows your history to resolve loading.

closed 2022-05-19 by NotExpectedYet