OctoFarm/OctoFarm #621: Printer profile PATCH in state.js updatePrinter does not send relevant profile data

davidzwa, 2021-06-07:
### Description of the problem

```bash
@NotExpectedYet I had to fix a bug in `statejs` I created. We're calling PATCH for the printer profile, but no data was sent as body. That was my doing, which is now fixed here.
My question to you is why the data we're sending to update the OP printer profile does not match the OP REST API? It does not do anything right now, because all the properties in the body are discarded by OP.
https://docs.octoprint.org/en/master/api/printerprofiles.html#update-an-existing-printer-profile

The code where the profile is patched, is found here:
https://github.com/OctoFarm/OctoFarm/blob/a11c54f7f2f9aca8fb7fea54b2af7ce106f84262/server_src/runners/state.js#L3371
```

### Reproduction of the problem

```bash
...
```

NotExpectedYet, 2021-06-12:
I don't get what you mean it doesn't match? it did work before...

NotExpectedYet, 2021-06-12:
That's not the right object. There should be another one there with the profile information.

NotExpectedYet, 2021-06-12:
```body: JSON.stringify({ profile: settings.profile }),```

Should be sending that, not the clean thing. Probably me not paying attention.

NotExpectedYet, 2021-06-12:
I don't even think ``` let cleanProfile = removeObjectsWithNull(opts);``` that is needed over the profile one at all. It sends all the values anyway.

davidzwa, 2021-06-13:
> I don't get what you mean it doesn't match? it did work before...

Just for purpose of 100% clarity, git blame shows that the `let cleanProfile = ...` variable is from 5 to 8 months ago ;) Just reporting it here as regression so we dont forget it.

NotExpectedYet, 2021-06-13:
Oh I know this is me don't worry 🤣

davidzwa, 2021-08-13:
@NotExpectedYet I will tackle this in the canary code and thus the issue is stale.

closed 2021-08-13 by davidzwa