OctoFarm/OctoFarm #630: Windows OS - Slow loading and response times

davidzwa, 2021-06-18:
### Description of the problem

```bash
### WINDOWS USERS ###
Dear Windows users, a fix has been found for slow loading times on Windows. We will fix this in the **release candidate** 1.1.14-rc1. This means it will be a while before it is stable.
I hope to release 1.1.14 soon, because it seems pretty stable.

Cause:
System information loading CPU etc extremely slowly on Windows.
```

### Reproduction of the problem

```bash
Install on Windows, run and see the SystemRunner taking ages to load system information.
```

davidzwa, 2021-06-18:
Clinic Flame report before
[image]
After removing system information
[image]

davidzwa, 2021-06-18:
The query times per info type. I've decided to drop cpu and osInfo
```
cpu: 2.079s
cpuSpeed: 0.216ms
cpuLoad: 0.248ms
memoryInfo: 89.856ms
uptime: 1.996ms
osInfo: 2.137s
systemProcesses: 344.157ms
fileSize: 90.77ms
```

After removing:
cpuSpeed: 0.482ms
cpuLoad: 1.51ms
memoryInfo: 93.6ms
uptime: 18.715ms
systemProcesses: 362.594ms
fileSize: 97.104ms

davidzwa, 2021-06-18:
cpu property is redundant. Removing that.

copc, 2021-06-19:
Can't wait it is so unusable right now for me.

davidzwa, 2021-06-19:
@copc we are nearing a **release candidate** hopefully today, maybe tomorrow. Not guaranteed to be stable, but a lot better. Just needs some real testing.

Updating can only be done with git pull manually as this will be marked as a prerelease and the updater wont pick it up.

NotExpectedYet, 2021-06-27:
> Can't wait it is so unusable right now for me.

It's on development if you fancy been a tester for it. It should work fine I think after the recent pushes.

closed 2021-07-11 by NotExpectedYet