OctoFarm/OctoFarm #640: Release checked doesn't notify when API limit reached

NotExpectedYet, 2021-06-20:
### Description of the problem

```bash
When you hit your API limit on github checks the updater just states the below: 

20/06/2021, 06:53:30 | ERROR | OctoFarm-SoftwareUpdateChecker | Cant check release as it was not fetched yet or the last fetch failed. Call and await 'syncLatestOctoFarmRelease' first.

This is a bit mis-leading as any attempts at calling this will result in the same message. 

The below message is returned in the githubReleases key and can be utilised: 
{
  message: "API rate limit exceeded for 86.172.184.162. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)",
  documentation_url: 'https://docs.github.com/rest/overview/resources-in-the-rest-api#rate-limiting'
}
```

### Reproduction of the problem

```bash
exhaust your github api call limit and see the message.
```

davidzwa, 2021-06-20:
We can hard code the releases

davidzwa, 2021-06-20:
Options:

cache the releases on octofarm.net as public API endpoint
cache the releases locally and fall back to hard-coded value to make sure 'something' is known

NotExpectedYet, 2021-06-20:
> Options:
> 
> cache the releases on octofarm.net as public API endpoint
> cache the releases locally and fall back to hard-coded value to make sure 'something' is known

I'd rather it be automated personally, don't mind either solution.