OctoFarm/OctoFarm #643: Add FarmPi host check

davidzwa, 2021-06-20:
### Description of the problem

```bash
Check the FarmPi version on /system page
```

### Reproduction of the problem

```bash
...
```

mkevenaar, 2021-06-20:
Check if the file exists. If it does not exist, should we ignore the whole section or show `unknown`?

davidzwa, 2021-06-20:
I was thinking of replacing the distro (Windows 10, Linux, or FarmPi). Does that have your fancy?

mkevenaar, 2021-06-20:
Something like this?

[image]

davidzwa, 2021-08-20:
Will be worked on, but tracked on DevOps.

NotExpectedYet, 2021-12-17:
[image]

Will be in V1.2 :) 

Just need to sort that extra space

closed 2022-05-19 by NotExpectedYet