OctoFarm/OctoFarm #647: Docker support for arm64

cranky-coder, 2021-06-24:
### OctoFarm Version

latest

### OctoPrint Version(s)

1.6

### pm2 or docker?

docker

### Operating System

Raspberry pi os 64, ubuntu arm 64

### Location of the problem

Not sure

### Description of the problem

```bash
Curious if the dockerbuild could be updated to include arm64 architecture.  Or if there is something that is preventing arm64 support.  I have a bunch of pi4's here running a kubernetes cluster and interested in seeing if i could get octofarm running in it.
```

### Steps taken to reproduce

```bash
Docker pull on pi os 64 doesn't run.
```

### What did you expect to happen

```bash
NA
```

### To avoid the problem I had to

```bash
NA
```

davidzwa, 2021-06-24:
Hi @cranky-coder, we're currently working on an image called FarmPi.
Visit the discord for the way to get a beta test download:
https://discord.gg/G6cCkqVV

For the rest we might start testing/emulating ARM in future as we saw that indeed the Raspberry Pi is a bit of a underdog right now for OctoFarm. This requires time which is precious. Does the raspberry pi image above act as an acceptable substitute for you for the meanwhile?

NotExpectedYet, 2021-06-25:
Did you try the alpine image?

davidzwa, 2021-07-04:
@cranky-coder we're awaiting your response.

cranky-coder, 2021-07-06:
On the dockerhub https://hub.docker.com/r/octofarm/octofarm/tags

the alpine image is listed as linux/amd64.  Not arm64.  So as far as I can tell, switching to that alpine image will still not give me the arm64 architecture docker support.

In the previous response, you ask "if the above raspberry pi image above act as acceptable substitute"  but I see no mention of pi image.  But I am looking for a docker image support on the arm64 architecture to deploy the octofarm control system into a kubernetes environment built on pi4 64bit modules.  That is why I was inquiring on the arm64 docker specifically.   A custom pi image at this time would not change anything.  I was just curious if the ci/cd process for the docker image could be adjusted to also add the arm64 instead of single arch.

Maybe with something like the docker buildx multi-arch
https://www.docker.com/blog/multi-arch-build-and-images-the-simple-way/

davidzwa, 2021-07-06:
A user posted this
https://github.com/OctoFarm/OctoFarm/discussions/573

cranky-coder, 2021-07-06:
That is for arm32.  Won't work a 64 bit raspberrry pi setup.  

Is the octofarm dockerfile available directly somewhere?  Maybe I can take a shot and building it for arm64 and do a PR

cranky-coder, 2021-07-06:
> That is for arm32. Won't work a 64 bit raspberrry pi setup.
> 
> Is the octofarm dockerfile available directly somewhere? Maybe I can take a shot and building it for arm64 and do a PR

NM.  I see it.  Ill see what I can come up with.

davidzwa, 2021-07-06:
> That is for arm32. Won't work a 64 bit raspberrry pi setup.

Have you tried it? Curious as I thought 32 bits is compatible with 64 (and 64 is not with 32 as we know).

cranky-coder, 2021-07-06:
So a small update.  I cloned the repo and was able to build a multi-arch image.  

Anyone wanting to test it https://hub.docker.com/r/crankycoder5/octofarm-multiarchtest  BUT.  I would advise against making this permanent.  

It was built of the monolithic dockerfile.  So, so far so good.  I am playing with it locally in my pi kubernetes environment.  

I am guessing you are using the workflows in the .github folder for the builds.  If you don't mind, Ill keep playing with it and see if by some chance I can get github to do a multi-arch build.  If I can, then I think that will give the multiarch builds automatically.

davidzwa, 2021-07-07:
Indeed we are using Github Actions (GHA) to build and push the images. We're open to adding a custom ARM based image if that's ok with you? It would involve adding a separate Dockerfile, a bit of instructions in `/docs/USING_DOCKER.md` and a github actions workflow file.
Fyi I was thinking of merging the docker jobs into 1 GHA yml file as the current way of releasing the docker containers isnt 100% to my liking. (Naming is wrong and sometimes certain PR's cause a docker build)

The monolithic approach is avoidable as there is a arm64v8 mongodb image.
https://hub.docker.com/r/arm64v8/mongo/ 

Interested to see your Dockerfile

davidzwa, 2021-09-14:
I have new information/script from building OctoPrint for RPi ARM v7 using Docker buildx, so we can give this a shot now.

hemsleyk, 2021-12-20:
Good to see this one in the pipe. I was able to download the zip and build the dockerfile on a Pi 3B+ without issues so having it "just work" via docker pull / run through multiarch will be great.

NotExpectedYet, 2021-12-23:
> Good to see this one in the pipe. I was able to download the zip and build the dockerfile on a Pi 3B+ without issues so having it "just work" via docker pull / run through multiarch will be great.

I've marked this as a good first issue / help wanted because I'm not sure when I'll get time to focus on docker more. Any help anyone wants to bring to the table would be appreciated as I've limited docker knowledge.

NotExpectedYet, 2022-01-31:
[image]

These are now building thanks to @jwkicklighter 

development builds on tag: dev. 

master will build once V1.2 hits

closed 2022-05-19 by NotExpectedYet