OctoFarm/OctoFarm #648: `serverSettings.influxExport.active` index can fail

davidzwa, 2021-06-24:
### Description of the problem

```bash
This is a startup order problem mostly apparent during tests. It will fix itself with a dependency injection system in V2. For now: move to `app-core`
```

### Reproduction of the problem

```bash
...
```

NotExpectedYet, 2021-06-27:
I believe I may have inadvertently fixed this in the settings PR. Not checked haven't been able to reproduce locally.

davidzwa, 2021-06-30:
> I believe I may have inadvertently fixed this in the settings PR. Not checked haven't been able to reproduce locally.

Personally I only see this in tests. It seems to be a race condition, or that's my escape argument for most of these things. I hope to move all startup calls to app.js or app-core.js one by one to see the impact

davidzwa, 2021-07-26:
Marking this as fixed as the server settings are super reliable on canary.

closed 2021-09-25 by davidzwa