OctoFarm/OctoFarm #656: Current Operations never loads on development dashboard

NotExpectedYet, 2021-06-27:
### Description of the problem

```bash
Current Operation panel doesn't load when opening dashboard, stays spinning
```

### Reproduction of the problem

```bash
Load dashboard. Views not affected.
```

davidzwa, 2021-06-27:
I think this has to do with the SSE bug => flatted PR, so not specific to current ops. Correct me if Im wrong.

NotExpectedYet, 2021-06-27:
> I think this has to do with the SSE bug => flatted PR, so not specific to current ops. Correct me if Im wrong.

Everything else loads, on every other page. Just contained to Current Ops. Didn't really look further into it like.

davidzwa, 2021-06-27:
For me nothing loads. Might be a good idea to check if your server node modules are up to date with the package lock.
Your printer management PR showed continuous spinner and errors server side for me

NotExpectedYet, 2021-06-27:
> For me nothing loads unless I fix the SSE flatted require path. Might be a good idea to check if your server node modules are up to date with the package lock.
> Your printer management PR showed continuous spinner and errors server side for me

Yeah thats probably right will check

davidzwa, 2021-06-27:
This could be a breaking change for users who forget to do npm ci - - production...

NotExpectedYet, 2021-06-27:
> This could be a breaking change for users who forget to do npm ci - - production...
> 
> Maybe do startup checks?

The updater should really do `npm ci --production`` now rather than npm install now that issue's all gone. It should check for out of date packages as well, rather than just missing packages. 

I mentioned a start up check for packages and you didn't like the idea?

NotExpectedYet, 2021-06-27:
I've upped in on the package.json, with the addition of the OF package the updater will run `npm install` before restarting now anyway so should be good.

davidzwa, 2021-06-27:
If I remember correctly, and that's pretty vaguely, I dont like the execSync commands. We've had a lot of bugfixing with those and parsing output/errors was a small hell. 
Also I don't like calling npm as its super slow and not JSON - Id prefer to call a rapid API for it instead of command line - (maybe yarn?)

Finally this flatted thing is about a breaking change, just like Node 12. Its something that would prevent normal usage of OctoFarm. If that's the kind of thing you had in mind, Im game. 

So in python we wouldn't need to call pip to get us the package version. We'd use the __version__ module attribute. Any way we have that here? My intention is to avoid NPM at all cost. (package.json or lock arent good enough sadly)

NotExpectedYet, 2021-06-27:
> If I remember correctly, and that's pretty vaguely, I dont like the execSync commands. We've had a lot of bugfixing with those and parsing output/errors was a small hell.

I don't remember it that way, I remember the small hell been my goofs not exec. That's not saying I wouldn't wanna move away from execSync mind. I just remember mentioning adding a check to startup it hadn't anything to do with execSync, anyway not an issue was just curious aha. 

> Also I don't like calling npm as its super slow and not JSON - Id prefer to call a rapid API for it instead of command line - (maybe yarn?)
> 

> Finally this flatted thing is about a breaking change, just like Node 12. Its something that would prevent normal usage of OctoFarm. If that's the kind of thing you had in mind, Im game.
>
> So in [...]

Well the only thing I can think of is parsing the package.json and comparing installed or getting the npm module to work inside the app. Just failed last time I tried but I wonder if it had a 14 limit? I'll revisit it, unless you wanna take a look?

NotExpectedYet, 2021-06-27:
> I think this has to do with the SSE bug => flatted PR, so not specific to current ops. Correct me if Im wrong.

This is still an issue for me even with the changes and updated packages. `npm outdated` doesn't show flatted been out of date.

davidzwa, 2021-06-27:
So to summarize my goof as communicated offline:
- SSE flatted is not the problem here, it was me being a stupido by not reinstalling my node_modules folder after switching from the OIDC branch. I ran node-check-updates there, ncu for short and bumped a lot of packages. Worth a look btw.
- the investigation for this issue is likely related to webpack PR... Id name change maybe or partial ejs change?
- title of this issue is incorrectly edited by me

NotExpectedYet, 2021-06-27:
You imported the file fine and the correct variable was going through. Think it would throw an error if not.

davidzwa, 2021-06-29:
For me this bug doesnt occur on dev

NotExpectedYet, 2021-06-30:
Cool will take another look.

NotExpectedYet, 2021-07-02:
So just to confirm this is not as the titled and actually the dashboard isn't reacting to the field and displaying CO when it's not enabled.

closed 2021-09-30 by github-actions[bot]