OctoFarm/OctoFarm #668: If the first timeout fails, the second doesn't seem to use the increase timeout value.

NotExpectedYet, 2021-07-02:
### Description of the problem

```bash
User couldn't establish a connection to printer after RC upgrade. 

Logs said: timeout retrying in 5000. 

The initial timeout of 1000ms didn't work, and probably never worked. So I gather the setting to get the secondary timeout value which defaults to 10 seconds never got updated on the connection attempt.
```

### Reproduction of the problem

```bash
...
```

NotExpectedYet, 2021-07-02:
```

  async getUsers(printer, retry = false) {
    return this.getWithOptionalRetry(printer, apiUsers, retry);
  }

```

That should be retry = true

davidzwa, 2021-07-02:
This was bound to happen once or twice. Good find!

NotExpectedYet, 2021-07-02:
> This was bound to happen once or twice. Good find!

GM pointed it out like, man steamed ahead live in production aha. He's also said the files on one of his printers isn't syncing but awaiting more info on that.

davidzwa, 2021-07-02:
By the by, interested if the timeout not increasing is the problem or not retrying.
As stated offline I dont think retrying would be the fix, slowly increasing timeout would.
And as also said offline, I hope to work towards a more robust timing manager solution

Have to investigate either way.

NotExpectedYet, 2021-07-03:
> By the by, interested if the timeout not increasing is the problem or not retrying.
> As stated offline I dont think retrying would be the fix, slowly increasing timeout would.
> And as also said offline, I hope to work towards a more robust timing manager solution
> 
> Have to investigate either way.

Yeah I think it's that, just investigating now and what I presumed above is wrong after looking at state.js. You'd already passed the true flag for the retry's where applicable, so it is retying and I'm seeing the timeout increase to +9000.

NotExpectedYet, 2021-07-03:
The fetchApiTimeout is being passed the secondary time out but it's not using it, it's still timing out at 1000ms.

NotExpectedYet, 2021-07-03:
Yeah was the global API check not have retry. The rest already did. Nothing wrong with your functions or the original. 

Like you say could do with something to actually cancel that but it's work as it was on that PR now so I consider this issue fixed. Wrote my investigation up on the PR and what additional cleanups I did from some personal goofs a while back.

closed 2021-07-11 by NotExpectedYet