OctoFarm/OctoFarm #672: Filament Temperature Offset not working

enderjones, 2021-07-03:
### OctoFarm Version

latest

### OctoPrint Version(s)

1.5.3

### pm2 or docker?

pm2

### Operating System

Ubuntu via Hyper-V

### OctoPrint Plugins

SMS notification

### Location of the problem

Not sure

### Description of the problem

```bash
I tried to put a -20 degC offset on one of my spools of filament, but it did not offset when I started the print.
```

### Steps taken to reproduce

```bash
1.
2.
3.
...
```

### What did you expect to happen

```bash
The temperature to be reduced by 20 degC
```

### To avoid the problem I had to

```bash
...
```

davidzwa, 2021-07-03:
Please provide an exact OctoFarm version because 'latest' is ambiguous as the 1.2 release candidate is published.

enderjones, 2021-07-03:
@davidzwa updated the version.

NotExpectedYet, 2021-07-04:
@davidzwa Its client side so unless there's any console logs happening when starting a print I'll just have to repro at minute. I know where the code is at a guess it's something not accounting for a (-) figure as this was fixed and tested a bit ago but only remember testing positives. 

Thanks for the report @enderjones I'll take a look and see if I can't get it fixed for rc2. Would you be able to test if so?

enderjones, 2021-07-04:
@NotExpectedYet Yeah for sure.  Thanks for looking into it!

enderjones, 2021-07-04:
Odd observation.  I started the print for the 3rd time and the temperature offset worked fine...  I did not change anything, just started the same file and it worked correct

NotExpectedYet, 2021-07-04:
Hmm well, I'll leave this open till I can actually investigate it more. I think I have an idea why anyway. 

I forget the conditions that are in that section of code but there's a fair few from tacking on features over the course of last year. I gather you managed to hit a condition I didn't update when testing it last time. More reason to clean that code up :grinning:

If you have any info to the steps you took and what might be different will help account to it in the future. 

Least it's now working

pat1121, 2021-08-17:
Just to add to this, I can confirm that the temperature offset:

1) Does NOT work if you print from the file manager (uses standard temps in Gcode)
2) DOES work if you print by selecting the printer in the printer menu, then selecting the file you'd like to print

I am using a +20 temp offset, haven't tried negatives.

NotExpectedYet, 2021-08-17:
> Just to add to this, I can confirm that the temperature offset:
> 
> 1. DOES work if you print from the file manager (uses standard temps in Gcode)
> 2. DOES work if you print by selecting the printer in the printer menu, then selecting the file you'd like to print
> 
> It was suggested it only worked from the printer menu, I got it to work from both.
> I am using a +20 temp offset, haven't tried negatives.
> 
> It did NOT work the first time I tried to print the file (on a few separate printers). I had to cancel (before it even started) the first print, then tell it to print again. On the second attempt, the temperature offset worked. This happened on 4 printers, 4 separate prints the exact same way, using either the printer menu or the file menu to start the print.

Thanks for that! Makes it easier to track down :)

pat1121, 2021-08-17:
On the second print (using the same printers after going through the start / cancel procedure above) everything works as intended using all 3 ways to print (reprint, printer menu, file menu) without having to do the start / cancel part.

NotExpectedYet, 2021-10-17:
Sorry about my tardiness with this issue but I've got it solved for v1.2. 

Works from any area in the UI with positive and minus values.

Normal: 
[image]
-25: 
[image]
+25:
[image]

It worked after the second one because I was incorrectly sending the offset after the start print command, so the offset had been applied by that time. Will update here once the fix has hit dev / master.

commit deccbe48 (2021-10-17): Fixed #672: Incorrectly sending spool temp offsets after print start not before
commit 93beffd9 (2021-10-17): Bug Fixes Galore (#760)

* update versions for 1.2 realease

* clean up

* further clean up

* further further clean upper

* octofarm client package fix

* fix incorrect client import

* fix no connection logs spinner

* update packages

* fix incorrect information been passed to error modal

* fix scan network for printers

* fix importing of printers and websocket issue

* fix cors check

* also remove ping message

* Printer Manager bulk commands update

* Printer Manager plugins + fix uniqueness

* Fix double listener issue, should also fix re-creating the connection with different API ke [... cut]
closed 2022-05-19 by NotExpectedYet