OctoFarm/OctoFarm #703: OctoFarm (Farmpi) not connecting/initializing printer

jaymcd, 2021-07-22:
### OctoFarm Version

1.1.13

### OctoPrint Version(s)

1.61

### pm2 or docker?

pm2

### Operating System

Farmpi RaspiOS

### OctoPrint Plugins

Octoscreen, UI Customizer, Astroprint, Printoid, etc

### Location of the problem

Not sure

### Description of the problem

```bash
...
Octofarm will scan and find my Octoprint server but always shows the printer as Shutdown and offline when it isn't. I can control the printer from Octoprint ( it is connected and showing an IP address). CORS is enabled, Application API setup. ??? Octofarm even seems to hang Octoprint ( or slow it down to nearly hanging. As soon as I remove the Octoprint server from Octofarm it seems to stop hanging and operate normally and respond at normal speed.
```

### Steps taken to reproduce

```bash
Deleted and re scanned for the printer. Added it manually even,
Restarted Octoprint a number of times. Tried with a different printer even. Same result.
```

### What did you expect to happen

```bash
...
I expected to see the printer online
```

### To avoid the problem I had to

```bash
...
I do not know. Is it possible that Octoscreen UI is interfering? I can try with it uninstalled.
```

NotExpectedYet, 2021-07-22:
Hi could you tell us a little more about your setup please? 

When you Scan for printers it expects and API inputting and I wasn't sure with your explanation. Did you create an Application Key on OctoPrint? 

Do you have multiple users on OP or just a single user? 

Could you upload a screenshot of what your seeing on the Printers page after adding the printers.

jaymcd, 2021-07-22:
I have Application Key created  and  CORS setup (noted above) - with just one user. Here is a screenshot or two 
[image]
[image]

Here is a full list of my plugins;
AstroPrint: AGPLv3
Autoselect Plugin: AGPLv3
Backup Scheduler: MIT
Bed Leveling Plugin: AGPLv3
Bed Leveling Wizard: AGPLv3
Bed Visualizer: AGPLv3
Calibration-Companion: AGPLv3
Change Filament Plugin: bsd-3-clause
Consolidate Temp Control: AGPLv3
Cost Estimation: AGPLv3
Cura Thumbnails: AGPLv3
CuraEngine Legacy: AGPLv3
Dashboard: AGPLv3
DisplayLayerProgress Plugin: AGPLv3
DisplayProgress: AGPLv3
Draggable Files: MIT
Dragon Order: AGPLv3
Filament Manager: AGPLv3
Filament Sensor Simplified: AGPLv3
FileManager: AGPLv3
Firmware Updater: AGPLv3
Fix CBD Firmware Plugin: AGPLv3
Google Drive Backup: MIT
GridSpace Plugin: MIT
ipOnConnect: AGPLv3
Marlin GCode Documentation: AGPLv3
MyMiniFactory: AGPLv3
OctoPod Plugin: Apache-2.0
OctoPrint-WiFiStatus: AGPLv3
Preheat Button: AGPLv3
PrettyGCode: AGPLv3
Printoid Plugin: Apache-2.0
PrintTimeGenius Plugin: AGPLv3
Resource Monitor: MIT
Sidebar Temp Graph: MIT
Simple Emergency Stop: AGPLv3
Slicer Thumbnails: AGPLv3
Stateful Sidebar: AGPLv3
STL Viewer: AGPLv3
Tab Order: AGPLv3
Tempsgraph Plugin: MIT
Terminal Commands Extended: AGPLv3
Touchtest Plugin: AGPLv3
UI Customizer: AGPLv3
USB Keyboard: AGPLv3
WebcamTab: AGPLv3
Z Probe Offset: AGPLv3
Z-Bolt OctoScreen: AGPLv3

NotExpectedYet, 2021-07-22:
> I have Application Key created and CORS setup (noted above) - with just one user. Here is a screenshot or two

Thanks for those, it seems it thinks your Pi is offline. 

I'm not aware of any bugs in that regard so will need a bit more information to hopefully get to the bottom of this. 

Would you mind doing a log dump for us from the system page and uploading it? Just to warn it may contain API keys and IP addresses of your device until we scrub those in a future release. 

I gather your running OctoPi on 192.168.1.229 just to confirm? Just want to make sure it's not on a weird port or anything like that. 

If you could try getting SSH access to the device too could you just confirm your FarmPi box sees the octoprint instances. Obviously your browser does like but I'd just like to rule out any potential network issue. `ping 192.168.1.229` in the command line will work. 

Something is making OF think your device is offline so hopefully will help us figure out what.

NotExpectedYet, 2021-07-22:
Just to add as well as I didn't mention it before but I don't think any of those other plugins would cause this issue. Nor OctoScreen although I don't personally use it.

jaymcd, 2021-07-22:
Scaning finds the instance though...

It seems like it is having trouble with API access

I will try a default octoprint setup and the latest nightly if Farmpi.

Jay

Jay M. McDonald

jaymcd, 2021-07-22:
Thanks for the quick response and help btw. I will try and reduce some variables. I have both ethernet and wireless network connectivity to the Octoprint device.  Farmpi is a Raspberry Pi400, the Octoprint Pi is a 3b+ btw. I can get ssh access to the Octoprint system on this network from all devices I have tried. Benn running this system for sometime before trying Octofarm. Will try it from Farmpi box after setting up nightly version.

NotExpectedYet, 2021-07-22:
Yeah that's actually a very good point! Can I blame the heat? Aha 
I would have thought if it were network then ssdp would fail also. 

Out of interest have you tried a user API key? Both should work mind. 

Sounds like it might be something with the API key as you said though. You could try running OP in safe mode as that would rule out a plugin issue without having to install a fresh system. 

Your welcome too! We will surely get to the bottom of the issue.

jaymcd, 2021-07-22:
new nightly - same thing but adds this message - 11:32:41 | https://192.168.1.229 | request to https://192.168.1.229/api/settings failed, reason: self signed certificate retrying in 5000

jaymcd, 2021-07-22:
same situation using a user api key. I will try creating an octofarm with a new key user and see what happens.

NotExpectedYet, 2021-07-22:
> new nightly - same thing but adds this message - 11:32:41 | https://192.168.1.229 | request to https://192.168.1.229/api/settings failed, reason: self signed certificate retrying in 5000

Should it be https? Or http? We don't currently support https with self signed certificates.

jaymcd, 2021-07-22:
Odd. I changed it to try it and then changed it back... Will see what it is doing with a different instance of octoprint

Jay M. McDonald

NotExpectedYet, 2021-07-22:
> Odd. I changed it to try it and then changed it back... Will see what it is doing with a different instance of octoprint Jay M. McDonald
> […](#)

That could be indeed a bug tbh. I noticed some when working through the SSL stuff myself at work. 

Deleting a printer should clear it if it isn't.

jaymcd, 2021-07-22:
tried a different instance of Octoprint - same issues. That said it has similar plugins. Going to try a different printer and see if there is a serial comm issue.

jaymcd, 2021-07-22:
same issue with a different printer. odd. I can ssh to this Octoprint instance from teh Farmpi system - no trouble. Must be some plugin causing an issue. I will try a new Octoprint install.

jaymcd, 2021-07-22:
ok. definitely something with the Octoprint setup / possibly plugins. Just setup a default Octoprint instance - connects fine. I guess I will disable plugins and add on UI stuff ( Octoscreen) one by one and see what's what.

jaymcd, 2021-07-22:
I disabled most of my plugins to no avail. I am not sure what the issue is. The problem happens on two different octoprint instances ( on separate SD cards) yet the default/clean octoprint install seems to work fine. Not sure what to say other than I will rebuild based on the clean install that I can get Octofarm to see.

NotExpectedYet, 2021-07-22:
> I disabled most of my plugins to no avail. I am not sure what the issue is. The problem happens on two different octoprint instances ( on separate SD cards) yet the default/clean octoprint install seems to work fine. Not sure what to say other than I will rebuild based on the clean install that I can get Octofarm to see. The connection to the fresh Octoprint install is almost instant so...

Hmmm that sounds like a plugin then but honestly couldn't be sure which without process of elimination. 

Not sure which ones you've tried but I'm happy to leave this open to learn more. 

At a guess: 
Astroprint 
IpOnConnect (Not sure what that is) 

I doubt OctoPod plugin and same with Printoid I've used the latter and have users with the former.

NotExpectedYet, 2021-07-22:
It's not a combination of OctoFarm + OctoScreen? I gather they've both got different application keys so shouldn't. 

If you can post OctoPrint logs and OctoFarm logs we can investigate further for you. Unless you know how to check them yourself you'll be able to see what's happening when OF tried to connect that way. 

online@notexpectedyet.com if you don't want to upload them publicly. Not sure what to suggest otherwise would just be guesswork.

jaymcd, 2021-07-22:
I will have to get at that later. The fresh install works and the process of adding plugins is awesome.

NotExpectedYet, 2021-12-18:
Going to close this as the original issue was resolved. Please do let me know if you discovered the plugin causing the issues, if not not worries!

closed 2021-12-18 by NotExpectedYet