OctoFarm/OctoFarm #710: Mon calling OP start doesnt check if file is selected

davidzwa, 2021-07-24:
### Description of the problem

```bash
BigOne branch
```

### Reproduction of the problem

```bash
...
```

NotExpectedYet, 2021-07-25:
The 3 printhead calls has been around for a while. 

This looks like a canary problem, can't repro on dev. "Print" button is correctly disabled without a file selected. There's no UI alerts because it should be disconnected, which leads me to think a state isn't been passed to the client. 

Doubt it's axios considering it's not even on the octoprint client yet

closed 2021-07-25 by davidzwa

davidzwa, 2021-07-25:
The reason for the OP API errors was the feedrate and flowrate factor body values were `null`. That's ofc related to me hacking together the backend with stub values.
Would be nice if the OP client validates required variables in future